realsenseai/librealsense #3098: realsense-viewer ros bag problem

Sun1307010608, 2019-01-18:
[FATAL] [1547818132.615885447]: Character [ ] at element [32] is not valid in Graph Resource Name [/device_0/sensor_0/option/Enable Auto Exposure/value].  Valid characters are a-z, A-Z, 0-9, / and _.

hello, everyone, I use realsense-viewer tool to record the realsense data, and the saved file is *.bag format file, I want to run this bag file by "rosbag play record_file.bag" and some errors occured, can anyone help me? thank you!

RealSenseCustomerSupport, 2019-01-28:
Hi Sun1307010608,

Please be aware that RealSense bag file is not compatible with what rosbag uses.
You may check out this sample for RealSense play/record.

Thanks!

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Hi Sun1307010608,

If there is nothing else, this will be closed soon.

Thanks!

closed 2019-02-11 by RealSenseCustomerSupport