realsenseai/librealsense #3099: How do I trace back the device from the frameset in the callback?

Wh1teSh4dow, 2019-01-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  4.7.0.intel.r5.0                                  | 
| Platform                                 | Skull Canyon  |
| SDK Version                            |  2.17.0                          | 
| Language                            |  C++                         | 
| Segment			|  Automotive                        | 

I have multiple D435i realsense cameras configured simultaneously with the same settings (streams, fps, resolution etc).
They are configured to be handled by a callback which receives framesets from either camera.

When I receive the frameset, I'd like to handle them differently based on which device they come from.

How can I identify which device provided with them from the frameset?

I don't see any identifiable information. Thank you!

dorodnic, 2019-01-22:
Hi @Wh1teSh4dow 
This is a gap in our API right now. For now, you could map the profiles you received from `pipeline.start()` with `frame.get_profile()` via `unique_id`. That said, I think it makes sense adding `get_sensor()` and `get_device()` API to `frame` for multi-camera use-case.

Wh1teSh4dow, 2019-01-23:
Yes, those enhancements were my thought as well. If it can't be done yet, you are right, the unique_id() function at least allows for differentiating of the different devices from which the frames/framesets come from, even if you can't tell specifically which one it is.
Thanks!

closed 2019-01-23 by Wh1teSh4dow

dorodnic, 2019-06-16:
As a follow-up, there is now `frame::get_sensor` method in newer releases