realsenseai/librealsense #3101: How to get pose data

MaoPann, 2019-01-20:
| Camera Model                       |D435i| 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            | 2.17.1  | 
| Language                            | c++   | 
| Segment			|  Robot  | 

### Issue Description
Hello, I want to get the angle quaternion through pose_frame of D435i, but when I call the get_pose_data() function, I encounter an exception. How can I solve it? 
[image]

ev-mp, 2019-01-20:
@MaoPann, the RS2_STREAM_POSE stream format is not part of D435i spec. 
You may run `rs-enumerate-devices` to obtain the list of all supported formats and configurations.

There is a demo being reviewed for motion tracking (pose) based on gyro and accel readings ([PR3053](https://github.com/IntelRealSense/librealsense/pull/3053)) .

Duplicated in  #3092,#3070 and #1078.

MaoPann, 2019-01-21:
ok,thanks

closed 2019-02-03 by RealSenseCustomerSupport