realsenseai/librealsense #3102: Cant run ./rs-depth with Android

cucrui, 2019-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  | 
| Operating System & Version |  Android  | 
| Kernel Version (Linux Only)    |  6.0.1 | 
| Platform                                 | Nexus 5X|
| SDK Version                            |  IntelRealsense 2.0 | 
| Language                            |  C++ | 
| Segment			|  Smartphone  | 

### Issue Description

I followed [this document](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md)
and generated binary file successfully.Then I connected Realsense with my Android phone, opened terminal and ran rs-depth binary file, met a problem "There are 0 connected Realsense devices." 
Is that something wrong?I have no idea about this problem.

RealSenseCustomerSupport, 2019-01-28:
@cucrui Which librealsense version are you using? Could you please check if the camera can be recognized by typing " l**s -al /dev/video*_ _**" in adb shell?  If the camera can't be recognized,  it might be USB OTG cable problem.  If change USB OTG cable can't work either, please attach adb logcat and dmesg for the problem. Thanks!

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@cucrui  Any update?

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@cucrui Will close the ticket as no update from you for a long time. If you still have question, please open another ticket. Thanks!

closed 2019-02-22 by RealSenseCustomerSupport