realsenseai/librealsense #3106: Texture alignment seems worse when align is used

spaltiel, 2019-01-20:
Windows 10
C++
D435

Dear Friends,

We continue to try and align texture and points cloud.
When using align it seems that texture is less aligned relative to without using it, also it seems that with align image texture is less smeared. 

the relevant code (with align):
		auto frames = pipe.wait_for_frames();
		auto processed = align.process(frames);
		rs2::depth_frame depth = processed.get_depth_frame();
                rs2::video_frame color = processed.get_color_frame();
		pc.map_to(color);

[image]

BTW the red object is approximately 0.5 meters away from the camera.

without align:
                auto frames = pipe.wait_for_frames();
		rs2::depth_frame depth = frames.get_depth_frame();
                rs2::video_frame color = frames.get_color_frame();
		pc.map_to(color);

[image]

Can you please help how to improve results so alignment will be correct and no smeared area will apear? What is the main cause for these issues?

Best Regards,
Amir I.

spaltiel, 2019-01-20:
> Aligning a 3D point cloud is a bit more complicated than a 2D image align,
> as you have to take the extrinsics of the camera into account to get
> correct results. Here is an example discussion of the subject:
>
> #2455 <https://github.com/IntelRealSense/librealsense/issues/2455>

Hi,

we followed the align example - is it not sufficiant?
what r u recommanding to do?

best regards,
A.I.

MartyG-RealSense, 2019-01-20:
I believe the alignment in the Align sample program is an alignment of 2D data and not a point cloud.

It is possible to align color and depth in the RealSense Viewer software instead of using scripting.  In its default 2D mode, set both the depth stream and RGB stream running.  Then click on the '3D' button in the top corner to enter 3D point cloud mode.

Using the lower of two drop down menus at the top of the Viewer screen in 3D mode, you can choose the color stream instead of the default depth stream setting, and the RGB then aligns to the point cloud.

Sorry I cannot provide an image to illustrate, I am away from computer on my tablet.

spaltiel, 2019-01-21:
10x!

we tried to add occlusion filter and the results are improved however yet the occlusions are still there.

Did you tried to rotate the RGB image based on the camera calibration and projecting the texture while taking into account the camera error?

anyway it can be nice solution.

Thanks and best regards,
Amir I.

spaltiel, 2019-01-21:
ev-mp,

thanks for your support!
it indeed helped us.

Amir I.

closed 2019-01-21 by spaltiel

ev-mp, 2019-08-04:
Duplicate of #2237, #2355