realsenseai/librealsense #3107: Abnormal angular velocity of gyroscope

MaoPann, 2019-01-21:
| Camera Model                       | D435i | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC|
| SDK Version                            |  2.17.1                      | 
| Language                            |  C++                      | 
| Segment			|  {Robot                   | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Get_motion_data obtains the angular velocity of the gyroscope. The angular velocity values of the gyroscope last night and this morning are quite different. Is that the reason for the camera itself?

ev-mp, 2019-01-21:
Hello @MaoPann , please provide details of the experiment you're running.
Additionally - what metric do you evaluate, is it the STD, mean/avg value or something else?

MaoPann, 2019-01-22:
@ev-mp Thank you for your reply .
I use RS2_STREAM_GYRO to get the angular velocity, and then integrate to get the final angular value. 
The results of two adjacent tests were similar. But testing after an hour will be quite different from the previous results. I print out the angular velocity of each frame and find that when rotated at approximately the same speed, the values of the angular velocity read will be quite different, resulting in the incorrect angle obtained by the final integration, so I think it may be a problem with the original data of realsense. For example, when the camera is still, the reading angular speed fluctuates between 0.1-0.5 for the first time and 1-5 for the second time. I don't know why. 
[image]

ev-mp, 2019-01-23:
@MaoPann , the gyro readings provided by Librealsense are `rad/sec`. So I presume you converted the 1-5 units to `deg/sec`, correct ?
The typical noise for gyro according to [BMI055 spec](https://www.mouser.com/ds/2/783/BST-BMI055-DS000-08-786482.pdf) is 0.1 deg/sec. 
`Output Noise  0.1 °/s ( rms, BW=47Hz @ 0.014°/s/√Hz)`

I would suggest to use `rs-data-collect` tool to validate the obtained results  [link](https://github.com/IntelRealSense/librealsense/tree/development/tools/data-collect#rs-data-collect-tool):
- The tool is capable of recording hours of IMU streaming (60 min of Gyro+Accel stream produce some 2.2 Gb text file). 
- The data is serialized in a convenient Excel-friendly csv format, ready to run statistics.

You can then share the data for review

closed 2019-02-11 by RealSenseCustomerSupport