realsenseai/librealsense #3109: Error calling rs_wait_for_frames ( device:0x1ef4130 ):  Timeout waiting for frames.

bf826989019, 2019-01-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.15)   | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.17.1}  | 
| Segment			|  {Robot/   | 

### Issue Description
I follow https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. And I can run realsense-viewer successfully. But when I roslaunch wrist_camera.launch. There are some problems. It show 

[ WARN] [1548052175.573291881]: /camera/driver - Detected unvalidated firmware:
				- 619204000646's current camera firmware is 3.21.0.0, Validated camera firmware is 3.10.10.0

and 

[ERROR] [1548052181.490165342]: /camera/driver - Error calling rs_wait_for_frames ( device:0x1ef4130 ): 
Timeout waiting for frames. 

I don't now what to do. 
Details are as follows：

... logging to /home/baifan/.ros/log/c9163d90-1d45-11e9-a47e-7c2a3166d0dc/roslaunch-baifan-TM1801-6024.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://baifan-TM1801:46575/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: SR300
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
 * /camera/driver/depth_fps: 30
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 480
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 640
 * /camera/driver/enable_color: True
 * /camera/driver/enable_depth: True
 * /camera/driver/enable_fisheye: False
 * /camera/driver/enable_imu: False
 * /camera/driver/enable_ir2: True
 * /camera/driver/enable_ir: True
 * /camera/driver/enable_pointcloud: False
 * /camera/driver/enable_tf: True
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/sr300_auto_range_enable_laser: 0
 * /camera/driver/usb_port_id: 
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)
  /
    convert_depth_to_m (nodelet/nodelet)
    realsense_link_broadcaster (tf/static_transform_publisher)
    standalone_nodelet (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/camera_nodelet_manager-1]: started with pid [6051]
process[camera/driver-2]: started with pid [6052]
process[camera/rgb_debayer-3]: started with pid [6053]
process[camera/rgb_rectify_mono-4]: started with pid [6054]
process[camera/rgb_rectify_color-5]: started with pid [6066]
process[camera/ir_rectify_ir-6]: started with pid [6078]
process[camera/depth_rectify_depth-7]: started with pid [6091]
process[camera/depth_metric_rect-8]: started with pid [6102]
process[camera/depth_metric-9]: started with pid [6115]
process[camera/depth_points-10]: started with pid [6132]
process[camera/register_depth_rgb-11]: started with pid [6143]
[ INFO] [1548052174.925630605]: Initializing nodelet with 4 worker threads.
process[camera/points_xyzrgb_sw_registered-12]: started with pid [6159]
process[camera/depth_registered_sw_metric_rect-13]: started with pid [6180]
process[standalone_nodelet-14]: started with pid [6197]
process[convert_depth_to_m-15]: started with pid [6212]
process[realsense_link_broadcaster-16]: started with pid [6230]
Intel RealSense F200_camera ; 2.60.0.0
Intel RealSense LR200_camera ; 2.0.71.18
Intel RealSense R200_camera ; 1.0.72.06
Intel RealSense SR300_camera ; 3.10.10.0
Intel RealSense ZR300_adapter ; 1.29.0.0
Intel RealSense ZR300_camera ; 2.0.71.28
Intel RealSense ZR300_motion_module ; 1.25.0.0
[ INFO] [1548052175.573183462]: /camera/driver - Detected the following camera:
				- Serial No: 619204000646, USB Port ID: 2-2-1-1, Name: Intel RealSense SR300, Camera FW: 3.21.0.0
[ WARN] [1548052175.573291881]: /camera/driver - Detected unvalidated firmware:
				- 619204000646's current camera firmware is 3.21.0.0, Validated camera firmware is 3.10.10.0
[ INFO] [1548052175.573365046]: /camera/driver - Connecting to camera with Serial No: 619204000646, USB Port ID: 2-2-1-1
[ INFO] [1548052175.749279013]: /camera/driver - Setting static camera options
[ INFO] [1548052175.749994844]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1548052175.753628355]: /camera/driver - Enabling Color in manual mode
[ INFO] [1548052175.753932089]: /camera/driver - Enabling IR in manual mode
[ INFO] [1548052175.754151342]: /camera/driver - Starting camera
[ INFO] [1548052175.758592717]: /camera/driver - Publishing camera transforms (/tf_static)
[ INFO] [1548052175.758706424]: /camera/driver - Setting dynamic camera options
[ERROR] [1548052181.490165342]: /camera/driver - Error calling rs_wait_for_frames ( device:0x1ef4130 ): 
Timeout waiting for frames. 

[camera/camera_nodelet_manager-1] process has finished cleanly
log file: /home/baifan/.ros/log/c9163d90-1d45-11e9-a47e-7c2a3166d0dc/camera-camera_nodelet_manager-1*.log
[camera/driver-2] process has finished cleanly
log file: /home/baifan/.ros/log/c9163d90-1d45-11e9-a47e-7c2a3166d0dc/camera-driver-2*.log

Can you help me? I am a tiro. Thank you!

RealSenseCustomerSupport, 2019-01-28:
----------------------------------------------

Freeman Lo, Jan 28, 11:08 PM

Hi @bf826989019,

What's your ros wrapper version? Is your librealsense version 2.17.1? Do you follow below instructions?
https://github.com/intel-ros/realsense

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi @bf826989019,

Is the issue clarified/resolved with the correct wrapper and librealsense? If no support required, will close this.

closed 2019-03-12 by RealSenseCustomerSupport

bf826989019, 2019-03-14:
I haven’t tried it yet. I will ask you in time after I have experimented. Thank you.

zxprog, 2019-10-02:
have you solved this problem? I also met this problem and had no thoughts