realsenseai/librealsense #3110: Could not find a version that satisfies the requirement pyrealsense2

duchengyao, 2019-01-21:
```
$ pip search  pyrealsense2
pyrealsense2 (2.17.1.457)  - Python Wrapper for Intel Realsense SDK 2.0.

$ pip install pyrealsense2
Collecting pyrealsense2
  Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2

```

Anaconda. Python3.7

lramati, 2019-01-21:
What is your OS/architecture?
Unfortunately, pip install is currently only available on Windows and select Linux architectures

ShikiGami, 2019-01-25:
I'm having the same issue with Windows 10 (1803) x86-64.

lramati, 2019-01-25:
Checking [pypi](https://pypi.org/project/pyrealsense2/#files), you can see that precompiled binaries are currently only provided for Python 3.6 and Python 2.7
You'll need to either downgrade your python environment or compile from source

RealSenseCustomerSupport, 2019-01-28:
Hi duchengyao,

Wonder if you followed what Iramati suggested and get some updates?

Thanks!

RealSenseCustomerSupport, 2019-01-28:
Hi ShikiGami,

Which version of python you have under Windows10 1803?
You may try what Iramati suggested as well.

Thanks!

ShikiGami, 2019-01-29:
>           
> Richard Ren, Jan 28, 9:28 PM
> Hi ShikiGami,
> I tried off a Windows 10 64-bit 1803, with a newer version of pip, "pip install pyrealsense2" works well.
> Thanks!

The problem seemed to be that I was using Python 3.7, and had to downgrade to use pyrealsense2.
I hope you add soon support for Python 3.7.
Thanks

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Hi ShikiGami,

Glad to know the issue is resolved. Please check back to the github release information for updates on python support.

Thanks!

closed 2019-02-07 by RealSenseCustomerSupport

LLUWE19, 2019-08-07:
Hi, hope everyone is well.

I had a similar problem as the original and followed similar steps to get pyrealsense2 to download using pip. E.g. I installed python 3.6 rather than 3.7 on my windows machine and then the pip install worked. 

a pip list shows:

```
certifi       2019.6.16
numpy         1.17.0
opencv-python 4.1.0.25
pip           19.1.1
pylsl         1.13.1
pyrealsense2  2.25.0.994
setuptools    41.0.1
wheel         0.33.4
wincertstore  0.2
```

Despite this I still cannot import the realsense2 and get the following error.

```
Traceback (most recent call last):
  File "testing.py", line 7, in <module>
    import pyrealsense2 as rs
ImportError: DLL load failed: The specified module could not be found.

```

Can anybody clarify this problem or have any suggestions as to what it might be? Previously I have also tried copying the pyrealsense2.pyd and realsense2.dll files to the Anaconda -> Lib -> site-packages folder but that gave the same error.

Any help greatly appreciated,

Thanks

I have also tried copying