realsenseai/librealsense #3112: Could not use CustomRM.exe

anranx, 2019-01-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |  win10  | 
| Platform                                 | PC |
| SDK Version                            | SDK2.0 | 
| Language                            | C++/C#| 

### Issue Description
Hi! I have two questions about the CustomRW.exe.
Q1:
I download the  Calibration Tool-2.6.8.0-Setup and CalibrationAPI-2.6.8.0-Setup, and installed to the C drive. I want to use the CustomRW.exe to read and write the calibration data, the CustomRW.exe just show the command window in a second then drop out directly. I checked out the installation environment in setup package/vc_redist.x64.exe, it show the error, because I have already installed...（I have VS2017, VS2013 on my computer)
I try to run the CustomRW.sln in the examples  package, it compiled successfully, but I connot choose the options like -r -g -w. When I press r, the program is quit.
**So how to use the CustomRW.exe correctly?** 
[image]

Q2:
**The reason why I need to use  CustomRW.exe is because I have disassembled the camera into depth module and Vision Processor D4 Board** . When I reassembled the camera, the intrinsic is already changed, so I need to recalibrate and write the new intrinsic into camera.
I didn't find the reference about recalibrate. I plan to do the following steps. 
Use the DynamicCalibrator to recalibrate, then write the .xml file get from DynamicCalibrator  into camera, so the intrinsic (FocalLengthLeft/Right, PrincipalPointLeft/Right) will be reset, right?

I find the defult resolution in DynamicCalibrator is 1280*800 in depth , 1980*1080 in RGB.
If I write the .xml file at this defult,specific resolution, the intrinsic at other resolution (424*240 )will be changed,too? Or should I change the resolution in DynamicCalibrator and calibrate again?(I did't find how to change resolution in DynamicCalibrator).
Anyway, how to reset the intrinsic after disassembled the camera?

RealSenseCustomerSupport, 2019-01-28:
----------------------------------------------

Freeman Lo, Jan 28, 10:57 PM

Hi @anranx,

1. If you successfully build the app, you should find and run "CustomRW.exe" in below location, right? If all required DLL files there and D435 camera connected, the tool with those options should run normally.
C:\CalibrationToolAPI\2.6.8.0\bin
2. Dynamic calibration tool is only for extrinsic calibration. Since you disassemble/reassemble the D435 camera, the OEM calibration is required. We have the OEM calibration tool provided for intrinsic/extrinsic calibration. But, the target chart in the below link is required.
https://click.intel.com/realsense-d400-camera-oem-calibration.html
More info about the custom calibration and custom calibration data access in below paper.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400%20_Custom_Calib_Paper.pdf

anranx, 2019-02-03:
@RealSenseCustomerSupport  Thanks for your answer! I am already solved the Q1, but I stilll have some question about my Q2.
1.Since both OEM Calibration and User Custom Calibration can calibrated the intrinsic and extrinsic, can I use the User Custom Calibration, which is in C:\CalibrationToolAPI\2.6.8.0\examples\CustomCalibration\CustomCalibrationTest.sln?
[image]
Actually,I didnt find the CustomCalibrationSampleApp.exe and 540x480_60mm.pdf. **The CustomCalibrationTest.sln and print-target-fixed-width.pdf are what the paper refers to?** 
I just find the OEM Calibration target is $1500. If the way I use the User Custom Calibration  with CustomCalibrationTest.sln and print-target-fixed-width.pdf is wrong, maybe buy another d435 is better choise.

2. The reason I want to disassemble the camare is I want to reduce the size. I find the Depth Module D430 is what I need. It is factory-calibrated? Is the effect of using the depth module as same as disassembling the camera?It says that a small change can lead to inaccurate calibration.
In another word.Whether the depth module or the module removed from the camera, small movement may result in incorrect calibration is their common problem?
[image]
[image]

3.I also have D435i, the depth and RGB camera is already factory-calibrated, so I only need do IMU Custom Calibration, right?

Looking forward to your reply and thanks for your answer!!

RealSenseCustomerSupport, 2019-02-13:
----------------------------------------------

Hi @anranx,

1. You can build the custom calibration sample app. Custom Calibration sample app requires a target chart "540x480_60mm" or a custom target chart you use (you need to modify the code in order to match your target chart).
2. "**print-target-fixed-width.pdf**" is used for dynamic calibration. You can find it in the installation folder "C:\CalibrationToolAPI\2.6.8.0\target".
3. If you disassemble it and take out D430 module, it could cause the module bending or flex. You need to re-calibration its intrinsic and extrinsic. In order to check the impact, you can also use our DQT tool to check its depth quality.
4. D430 module is another product which has no RGB camera integrated and the support stiffness. The module is factory-calibrated. You can purchase it in below.
https://www.mouser.com/Intel/Embedded-Solutions/Video-Modules/RealSense-D400-Series/_/N-8nl98?P=1y9bmz0Z1yy6lwu
5. You need to calibrate the IMU of D435i by yourself. A python script in librealsense is provided for that.

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi @anranx,

If no further question, will close this.

closed 2019-03-17 by dorodnic