realsenseai/librealsense #3114: How do I use RGB camera of D415?

czjczjczj1, 2019-01-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0                           | 
| Language                            |  C                        | 
| Segment			|  others                         | 

### Issue Description
Hello, I would like to use D415 to detect an Aruco marker and I already have the source code provided by OpenCV. I tried using a normal webcam to do this and it succeeded. So now I would like to use D415 and my question is how can I use the RGB camera of D415? Thanks for any help.

vladsterz, 2019-01-22:
You can check the provided examples.

dorodnic, 2019-01-22:
Hi @czjczjczj1 
I'm not sure about the built-in OpenCV APIs (I think there is a REALSENSE flag in CMake, but did not try it myself).
We do provide couple of OpenCV+RealSense examples - [wrappers/opencv](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv)

closed 2019-02-11 by RealSenseCustomerSupport