realsenseai/librealsense #3116: Trimodal distribution of depth readings for same scene

ajprax, 2019-01-22:
*Camera Model*: D415
*Firmware Version*: 5.10.13
*Operating System & Version*: MacOS 10.14.2
*Platform*: PC
*SDK Version*: 2.2.17
*Language*: python
*Segment*: Robot

### Issue Description
When reading depth for the same pixel in many consecutive  frames, the values for a trimodal distribution, with most values clustering near some center which seems to be most accurate, but two smaller modes centered around values fairly far from the "true" center. When reading a depth closer to the camera, the distance between mode centers is smaller. It seems like maybe this is the result of some discretized component of the depth reading jumping values, but in the relatively small scale at which I'm operating, the ~4-5mm mode jumps shown in the chart below represent a pretty substantial error. At times there may be a hundred or more consecutive frames outside of the center mode, which makes it hard to combat with any kind of temporal filtering. When this jump happens, all depth values in the image seem to jump in the same direction, though the relative depths are not quite preserved, as it seems like closer objects don't move as far.

For the moment I have a workaround by establishing the a "fixed point" somewhere in my frame and manually collecting a baseline measurement and discarding frames whose reads for that pixel fall outside of the center mode. Unfortunately at times this means I may go several seconds without a frame. Is there a clear explanation for why this happens, or how to prevent/mitigate it?

10000 reads of the same pixel in the same scene, less some ~900 which read 0:
[image]

vladsterz, 2019-01-22:
ehm wow. I've conducted a similar experiment once and I got a different result. My distribution was more Normal-like. Do you have any  fluorescent light near you camera?

ajprax, 2019-01-22:
My setup is in a typical office, so the lights are probably fluorescent (though I can't see them directly). Do you think the issue may be caused by the flickering? I can try drowning out the overhead light with an LED bulb today and see if that helps.

ajprax, 2019-01-22:
Adding an LED desk lamp didn't help, but I can't say reliably that it doesn't have a flicker also.

vladsterz, 2019-01-22:
It's is in fact a very interesting case. Hope I'm gonna find some time tomorrow to reproduce this.

ajprax, 2019-01-23:
No, my camera is mounted facing down at a table and the sample point is in the middle of a flat area with no objects in it. I have seen this same behavior with pixels near*er* the edge of objects (really just smaller objects with sharp drops on the sides), but haven't looked specifically at boundary pixels.

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

XiangyuX Sun, Jan 30, 6:04 PM

@ajprax Is the table reflected? Could you please provide a picture of your setup which will be more helpful to understand the problem?
And if you're looking for depth quality evaluation, please refer to https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_DepthQualityTesting.pdf

ajprax, 2019-02-01:
Things have moved around enough that I can't get a picture of what the setup was like when I captured this data. Previously we ran into some issues with reflections interfering with the IR pattern so the table is covered in paper. The sample point was on the paper. Could reflections from other parts of the scene interfere?

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@ajprax Any update from your side? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@ajprax  Could you please update? Thanks!

ajprax, 2019-02-15:
Hi, sorry about the delay. I've moved on to other parts of my application for now, with a workaround to detect and discard frames outside of the center mode. Time permitting I'll come back and try to help find a better answer to what was happening here, but I'm pretty swamped at the moment so I don't know when that will be.

I will say that I am still seeing this issue and that occasionally I have to discard (sometimes hundreds) of frames looking for a good one.

These were taken today.

A good frame:
[image]
A discarded frame:
[image]

Since I can't devote more time to this right now, if you want to close the issue to keep things tidy, that's fine with me.

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@ajprax Thanks for the update! Any light condition change during your capture? If we can have IR images and RGB images that will helps. Let's work on it after you can get back to it. Thanks!

closed 2019-02-25 by RealSenseCustomerSupport