realsenseai/librealsense #3118: Need Hardware Advice

MaryBD, 2019-01-22:
Hi, I need a longer USB cable for my D435 realsense camera, is there any suggestion from the company itself? Can I simply use any other usb 3 c type? 
Is there any trolley for this camera for outdoor movements? 
Thanks

MartyG-RealSense, 2019-01-22:
You should be careful when choosing a third-party USB-C cable, as these cables can have quality issues.

My personal recommendation for long cables for RealSense 400 Series cameras is Newnex.  They supply industrial grade USB equipment that is proven to be able to be extended up to 100 m using fiber cable and extender equipment.  

https://www.youtube.com/watch?v=GLQgR1jT04M

For outdoor use, you could use a wheeled tripod called a tripod dolly.  The 400 Series camera has a standard size screw thread for third-party tripods on its base.

MaryBD, 2019-01-22:
Thank you for your prompt advices. The tripod dolly seems quite handy but I also need to fix my laptop on it as well. 
Have you also tried this fascinating angle connector one? 
[image]

MartyG-RealSense, 2019-01-22:
I haven't tried an angled connector like that, though it should be able to work with the camera if the cable is of sufficient quality.

USB cables can also come in the form of ones with screw-locks on the connector.  These can be screwed into a pair of holes either side of the USB port on the side of the camera.  An advantage of these in applications where the camera is experiencing motion or vibration is that they can keep the cable firmly plugged in and prevent it from working its way out of the port during motion / vibration and causing a disconnection.  Here's an example of a screw-lock USB cable:

[image]

Regarding the need to carry a laptop, perhaps a tripod like this would meet your needs:

https://www.amazon.co.uk/walimex-Laptop-Projector-Pallet-Tripods-x/dp/B001T9J9JE/ref=sr_1_62?ie=UTF8&qid=1548162076&sr=8-62&keywords=camera+tripod+dolly

Another way to carry a laptop around with the camera is to wear a backpack and have the laptop in the backpack whilst the camera is on the tripod.

closed 2019-02-13 by RealSenseCustomerSupport

MaryBD, 2019-02-14:
Thanks for the advices, already bought and works quite well.

npd26, 2019-05-09:
@MartyG-RealSense  is there are any other options because this cable is too expensive. can we use USB sever over Ethernet with realsense d435?

MartyG-RealSense, 2019-05-09:
If you are referring to Newnex cables, an alternative supplier recommended by a RealSense user is OCP.

https://www.ocp.com/product/62-00214/

Yes, the 400 Series cameras are usable with ethernet.  An Intel employee developed a project called EtherSense.

https://github.com/krejov100/EtherSense

The CONIX Research Center at Carnegie Mellon also developed an ethernet-based solution for stitching together point clouds from multiple 400 Series cameras.

https://github.com/conix-center/pointcloud_stitching

ssb6096, 2024-10-11:
> You should be careful when choosing a third-party USB-C cable, as these cables can have quality issues.
> 
> My personal recommendation for long cables for RealSense 400 Series cameras is Newnex. They supply industrial grade USB equipment that is proven to be able to be extended up to 100 m using fiber cable and extender equipment.
> 
> https://www.youtube.com/watch?v=GLQgR1jT04M
> 
> For outdoor use, you could use a wheeled tripod called a tripod dolly.
> 
> The 400 Series camera has a standard size screw thread for third-party tripods on its base. For example, Intel has tested with a third-party tripod from Manfrotto.
> 
> [image]
> 
> [image]

Could you please recommend some tripod ?

MartyG-RealSense, 2024-10-11:
RealSense 400 Series cameras have a standard **1/4-20** size tripod screw hole on their base. For example, in the image above Intel tested with a tripod from Manfrotto.

You can find compatible tripods by searching stores such as Amazon for the term **1/4-20 tripod**