realsenseai/librealsense #3119: Mac OS X: XCode build failed due to not found usb-1.0

kvbogdanov, 2019-01-22:
Trying to build librealsense under Mac OS x 10.12.3.
Get error after "Product->Build" in XCode:

> ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)

I saw  #2048 issue, but solution doesn't work for me.
libusb actually installed:

> bash-3.2$ brew ls libusb
/usr/local/Cellar/libusb/1.0.22/include/libusb-1.0/libusb.h
/usr/local/Cellar/libusb/1.0.22/lib/libusb-1.0.0.dylib
/usr/local/Cellar/libusb/1.0.22/lib/pkgconfig/libusb-1.0.pc
/usr/local/Cellar/libusb/1.0.22/lib/ (2 other files)
/usr/local/Cellar/libusb/1.0.22/share/libusb/ (16 files)
bash-3.2$ ls -la /usr/local/lib | grep usb
lrwxr-xr-x    1 konstantin  staff        54 22 янв 18:16 libusb-0.1.4.dylib -> ../Cellar/libusb-compat/0.1.5_1/lib/libusb-0.1.4.dylib
lrwxr-xr-x    1 konstantin  staff        46 22 янв 17:55 libusb-1.0.0.dylib -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.0.dylib
lrwxr-xr-x    1 konstantin  staff        40 22 янв 17:55 libusb-1.0.a -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.a
lrwxr-xr-x    1 konstantin  staff        44 22 янв 17:55 libusb-1.0.dylib -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.dylib
lrwxr-xr-x    1 konstantin  staff        44 22 янв 18:16 libusb.a -> ../Cellar/libusb-compat/0.1.5_1/lib/libusb.a
lrwxr-xr-x    1 konstantin  staff        48 22 янв 18:16 libusb.dylib -> ../Cellar/libusb-compat/0.1.5_1/lib/libusb.dylib

kvbogdanov, 2019-01-22:
Found solution by myself:

Just do 
`/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`
in terminal before launching XCode

BUT!

My cam (**D435i**) not recognized by the viewer anyway

[image]
[image]

closed 2019-01-22 by kvbogdanov

peymon, 2019-01-23:
Thanks for the 
`/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`
Solution