realsenseai/librealsense #3124: D435 shows up as D430

drhalftone, 2019-01-23:
I just received my D435, and the camera has a sticker on it that clearly says D435.  I plugged it and opened the viewer which told me to update the firmward to 5.11.1.  So I did that.  I now run the viewer, and the camera comes up as the D430.  There is no RGB sensor.  Also, the camera occasionally comes up as USB 2.1 instead of 3.2.  In that second issue, I just unplug and replug the cable in until it comes up as USB 3.2  But I can't get the camera to show up as a D435 and having an RGB sensor.

I am running version 2.18.0 with firmware 5.11.1 on Mac OS X 10.14.2.

MartyG-RealSense, 2019-01-23:
The D435 camera has a D430 camera module inside it, which does not have an integrated RGB sensor (only the D415 module has a built-in RGB sensor).  In the D435 USB camera module, RGB support is achieved by separately attaching an RGB sensor to the 'Vision Processor D4' circuit board inside the casing.

If the RGB sensor inside the D435 casing cannot be detected then the camera is ID'd as being an RGB-less D430 model.

There was a similar case to yours 11 days ago over on the Intel Support website, where a user purchased a batch of four D435s and only one of them was detected as a D435.  There was another case before that back in March 2018.

After a very long discussion, another user who had experienced the problem posted a workaround that had restored their cameras to being detected as D435s.  It involved downgrading the firmware to an earlier version and then re-upgrading it a current version.

https://forums.intel.com/s/question/0D70P000006BWKySAO

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Will try and see how to deliver the video tutorial. Thanks.

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Is this issue fixed after firmware downgrading ? if the RGB sensor is no longer recognized, it might be the flex cable problem - could be disconnected although this kind of chance is very low.

Thanks
p.s. please ignore the previous reply.

drhalftone, 2019-02-06:
It worked exactly as prescribed. I downgraded and then upgraded.  All is well.

closed 2019-02-11 by dorodnic

LostCanadian, 2019-02-15:
Is it possible to re-open this case? I believe this is a bug. I have seen this with a couple of sensors - one was a D435 that started showing up as a D430, and the other is a D415 which showed up as a D410. I did try the workaround suggested in this case, which is to downgrade to an older firmware, then upgrade to the newest, and that seemed to work for the D415, but it didn't work for my D435. I can keep downgrading/upgrading to see if it eventually works, but obviously this is not something that should be happening in the first place.

It seems like something is screwy in either the driver or the firmware. Either way, it seems like this is occurring for multiple people and should be logged as a bug.

MartyG-RealSense, 2019-02-15:
To re-open the case, please do so with a button underneath the box where you write a comment, next to the green 'Comment' button.

It would not be recommended to keep downgrading the firmware.  After 20 downgrades, the camera will lock out further downgrades and you will only be able to change to newer firmware versions from that point onward.

You should get a further response from one of the Intel team members at some point after you re-open the case (I do not work for Intel).  Good luck!

LostCanadian, 2019-02-19:
"To re-open the case, please do so with a button underneath the box where you write a comment, next to the green 'Comment' button."
...I don't have that button. Maybe it's only visible if you opened the case originally.

"After 20 downgrades, the camera will lock out further downgrades and you will only be able to change to newer firmware versions from that point onward."
Well, this is good to know...but also seems a bit arbitrary. Why 20? Why have this at all? This probably isn't an issue for most people, but if we have to keep downgrading/upgrading in order to try and work around this bug, it will become a showstopper for us pretty quickly!

MartyG-RealSense, 2019-02-19:
Yes, I believe the button only appears if you were the creator of the case.  My apologies, I got your name mixed up with Dr Halftone, who created the case.

The 20 downgrade limit dates back to the original RealSense camera models of several years ago.  I cannot recall the exact official reason for it, but my memory is that it was a defensive measure against malicious hackers or malware repeatedly accessing the camera and altering its firmware (perhaps to take control of the camera and spy through it).

bensonhero, 2019-02-24:
I got my new D435 yesterday and I had the same issue as well. The realsense viewer keep showing D430 instead.

I have tried downgrade and upgrade the firmware ( 5.11.1.100 the latest one )a couple of times, but the viewer still show D430.

[image]

MartyG-RealSense, 2019-02-24:
Does it still show as D430 if you click on the 'More' button at the top of the RealSense Viewer options panel and select 'Hardware Reset', please?

[image]

bensonhero, 2019-02-24:
Yes, It still show D430. I have tried that before and It still won't fix the problem. is it possible that the hardware has something to do with it?

MartyG-RealSense, 2019-02-24:
As @RealSenseCustomerSupport said in an earlier comment in this discussion, this issue can occur if the RGB sensor's cable inside the casing has become disconnected.  This is a rare cause but has been known to happen.  Since opening the case yourself to attach the cable would invalidate the sales warranty, your only course of action would be to return the camera for a refund or replacement.

If you purchased the camera from Intel's online Click store, the link below gives details about the refund / replacement claim process.

If you did not purchase the camera from the online Click store then it should be returned to the place where you purchased it.

The Click return policy for each camera model can be found on that model's Click store listing under the 'Warranty and customer support' tab.  Below is the terms for the D435.

[image]

bensonhero, 2019-02-24:
I just asked my retailer for a replacement, hope the new one will work . Thanks for the help

bensonhero, 2019-02-25:
I got my replacement and It work FINE!
Thanks for the help again .

MartyG-RealSense, 2019-02-25:
Awesome news!  I wish you the best of luck with your explorations of RealSense.  :)

gayanbrahmanage, 2019-06-09:
The RGB sensor is not built in the D435. It is a separate sensor that connected to the D4 vision sensor.  So,  there is a chance to get disconnected or loose connected RGB sensor-cable from the main board with a falling down to the ground. If it is disconnected the D435 is detected as D430.  

First, try to Update or downgrade the Firmware. If it doesn't work, open the casing and check the connection. Then upgrade the Firmware (v5_10_6 tested and
[image]
 works).

MartyG-RealSense, 2019-06-09:
Thanks for your detailed guide, @gayanbrahmanage !

It should first be pointed out that opening one of the cased RealSense cameras instantly invalidates its sales warranty, preventing return / refund.  But if you don't mind about that, there's nothing to stop you from opening it.

Also, advice was recently given not to do firmware downgrade-upgrade at the moment if you can avoid it, as there is a risk of it causing problems.  That advice has not been withdrawn yet as far as I know.

https://github.com/IntelRealSense/librealsense/issues/4038?language=en_US#issuecomment-495800286

Thanks again gayan!

dherraez, 2020-03-20:
Hello everyone, I´m new here and I´ve been facing some issues upgrading and downgrading the camera and after some investigations I find this here: 

"After 20 downgrades, the camera will lock out further downgrades and you will only be able to change to newer firmware versions from that point onward."

Well, we need to change lot of times the FW version, older and newer versions. The FW downgrade stucks in a moment and it was not possible to do it (then I found the message above). Finally I upgrade to the newest version and afterwards I was able to downgrade. Does this means that the "Downgrade count" starts from 0 in that moment? Anyone knows a workarround for this? To be able to upgrade and downgrade between 3 different versions with no limit?

Thanks in advance!!!

MartyG-RealSense, 2020-03-20:
As far as I know there is a finite limit of 20 downgrades that is not changeable.  This has been in place throughout the history of RealSense cameras since 2015.  

I have not seen an official explanation for it but I speculate that it is a security mechanism to limit the potential for hackers and ill-intentioned programs such as "malware" to repeatedly alter the firmware to introduce security breaching routines.

If you would like to make an official request for a change of this policy, please create a new case by visiting the front page of the forum and clicking the New Issue button.  Then include the words 'Feature Request' in the title of your message.  I can then label it as a feature request so the RealSense team can track it.  Thanks!

https://github.com/IntelRealSense/librealsense/issues

dherraez, 2020-03-31:
@MartyG-RealSense Thank you very much for your support!! 

Just info: I was not able to downgrade the FW version, but I tried upgrading to the latest one (even that is not the one we are using) and it works. Then, I was able again to downgrade to the expected one. So, it is a problem how it works, but if someone has this limitation, this is a way of workarround. I didn´t try to down/upgrade again, so I dont know the number of retries left and I was afraid of getting stucked again... ;)

Thanks again @MartyG-RealSense 

Regards!!

MartyG-RealSense, 2020-03-31:
@dherraez   I'm very glad you found a solution to your problem - thanks so much for sharing the details with the RealSense community!   :)

cwule, 2020-04-21:
> 
> 
> The RGB sensor is not built in the D435. It is a separate sensor connected to the D4 vision processor. So, there is a chance to get disconnected or loosely connected the RGB sensor-cable from the main board with a large shock or falling down to the ground. If it is disconnected the D435 is detected as D430.
> 
> First, try to Update or downgrade the Firmware. If it doesn't work, open the casing and check the connection. Then upgrade the Firmware (v5_10_6 tested).
> 
> [image]
> 
> [image]

Just a heads up that opening the case and disconnecting and reconnecting the cable solved the problem for me.

MartyG-RealSense, 2020-04-22:
@cwule   Thanks for the images!  Yes, there have been a few cases where the RGB camera cable in the D435 has come loose and made the camera be recognized as a D430 by the firmware.  As opening the case invalidates the sales warranty though, it is only worth doing if the camera was purchased more than 30 days ago so that it can no longer be returned for a refund / replacement, or if returning it to the place of purchase would be too difficult.  Thanks again!   :)