realsenseai/librealsense #3126: pipe.stop() after set_real_time(False) freezes the program when capturing from a rosbag file

mcasl, 2019-01-23:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  rosbag source  from D435                    | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10                                                 | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                  | PC                                                           |
| SDK Version                            |   2.17.1.457                                            | 
| Language                               | python                                                     | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
After opening a Rosbag file with real_time = False, stopping the pipe using pipe.stop() freezes the program.
Same code with real_time = True works flawlessly and stops the pipe.

This code freezes the program:
```
import pyrealsense2 as rs

filename = 'D:\\video.bag'
real_time = False

pipe = rs.pipeline()
cfg  = rs.config()
cfg.enable_device_from_file(filename)
profile  = pipe.start(cfg)
device   = profile.get_device()
playback = rs.playback(device)
playback.set_real_time(real_time)

pipe.stop()
```

This code works well, the only difference is the value of real_time:
```
filename = 'D:\\video.bag'
real_time = True

pipe = rs.pipeline()
cfg  = rs.config()
cfg.enable_device_from_file(filename)
profile  = pipe.start(cfg)
device   = profile.get_device()
playback = rs.playback(device)
playback.set_real_time(real_time)

pipe.stop()
```

This code also freezes the program, the only difference is setting real_time a second time:

```
filename = 'D:\\video.bag'
real_time = False

pipe = rs.pipeline()
cfg  = rs.config()
cfg.enable_device_from_file(filename)
profile  = pipe.start(cfg)
device   = profile.get_device()
playback = rs.playback(device)
playback.set_real_time(real_time)

playback.set_real_time(True)

pipe.stop()
```

lramati, 2019-01-30:
Is this behavior consistent for you? I ran the second snippet in a loop and couldn't get the program to freeze

mcasl, 2019-02-05:
It's actually the first snippet the one that freezes, not the second. In the second I just change the real time option to True to point out that it works then.

mcasl, 2019-02-07:
Possibly related to: https://github.com/IntelRealSense/librealsense/issues/2472

dorodnic, 2019-02-24:
@mcasl - I can confirm this is a bug. 
For now I work-around #2472, since we need to think how to handle this properly.

dmandrioli, 2019-03-05:
The problem occurs erratically in 2.18.1, it's really annoying for production code, I need to kill the process from outside and setting real_time to True is not a workaround since you can loose frames to process. @dorodnic Do you have an actual workaround for this? Thanks.

dmandrioli, 2019-03-05:
Well, a working approach is to postpone the processing after caching all frames in memory in real_time mode. Not so annoying :)

soarwing52, 2019-04-24:
using the lastest 2.21 version and still getting the freeze situation.

krejov100, 2019-04-30:
I am also facing this issue in 2.21

upxela, 2019-06-17:
facing this in 2.23 with both python and matlab.

kevindehecker, 2019-07-27:
Me too 2.24.

kevindehecker, 2019-10-01:
I just updated everything, now running kernel 5.0.0-29-generic with realsense api v 2.29, and this problems seems to be fixed!

closed 2019-10-16 by dorodnic