realsenseai/librealsense #3128: UI Offset Detected

smudgy67, 2019-01-23:
| | |
|---|---|
|**librealsense**|2.18.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense SR300|
|**Serial Number**|626204004015|
|**Firmware Version**|3.21.0.0|

Please provide a description of the problem

[image]

Hey there,

Each time I open the RealSense Viewer app, I get the display message as shown in the image.

On following the link for for the Intel Driver: Intel® Graphics Driver for Windows* [15.60], it doesn't state my Intel CPU version: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz, 3600 Mhz, 8 Core(s), 16 Logical Processor(s).

I have an NVIDIA RTX2070 graphics card installed.

Please can you assist? Thank you.

Kind regards,

Dominic

dorodnic, 2019-01-23:
Hi @smudgy67 
Sorry, this is obviously a false alarm. Please ignore the popup for now, we will try to fix this.

smudgy67, 2019-01-23:
Thank you Sergey

closed 2019-02-06 by RealSenseCustomerSupport