realsenseai/librealsense #3129: Announcement: Introducing the RealSense Tracking Camera T265

MartyG-RealSense, 2019-01-23:
Hi everyone,

A new RealSense camera product called the RealSense Tracking Camera T265 is now listed for pre-prder on Intel's online Click store, with pre-orders due to ship in the week of March 4 2019.

"Intel RealSense Tracking Camera T265 is a new class of stand-alone Simultaneous Localization and Mapping device, for use in robotics, drones and more".

https://click.intel.com/order-intel-realsense-tracking-camera-t265.html

barnjamin, 2019-01-23:
Thanks, Specifications table is a little light on details.  Any chance for some numbers like fps and depth accuracy?

dorodnic, 2019-01-23:
Thanks @MartyG-RealSense 

@barnjamin, more official information should be released today. 
Also, I'll try to answer some of the questions here.

[image]

## FAQ

1. **Is this a Depth-Camera?**
No, the T265 does not offer any Depth information. It can provide IMU (Accel and Gyro), two monochrome fisheye streams, and 6-DOF pose data.

2. **Can I use it together with a Depth-Camera?**
Yes, this device can be used with RealSense (and most likely any 3rd-party) depth cameras. It is passive and comes with an IR-cut filter.

3. **What platforms are going to be supported?**
At launch the T265 will be supported via librealsense on Windows and Linux, and via [ros-realsense](https://github.com/intel-ros/realsense). 
Mac-OS and Android support is planned but not scheduled yet, as well as OpenVR integration. 

4. **What are the system requirements?**
Any board with USB2 capable of polling 6-DOF packets 260 times a second should be sufficient. The device was validated on Intel NUC platform. 

5. **Will the same SLAM be available for the D435i as a software package?**
We are still looking into it, please stay tuned.

sam598, 2019-01-23:
@dorodnic Will the pose timestamps be able to sync with realsense depth cameras? Or will there be some method of synchronizing the pose data with other hardware?

dorodnic, 2019-01-23:
@sam598 - As far as I know there is no hardware-sync, but you could query T265 and D400 clocks with ~2-4ms accuracy each.

JBBee, 2019-01-23:
This product looks amazing.  Does it provide any support for persistence?  Ideally, a way to fetch and save the map on a host and then restore it and re-localize to that map?

Thanks.

dorodnic, 2019-01-23:
@JBBee - right now `save/load_relocalization_map` is still not part of librealsense API, but it is part of device capabilities and we are planning to support it at product launch in a months or so.

dorodnic, 2019-01-23:
Appreciate the feedback. 
Since this device is based on Myriad VPU, we might be able to add more CV functionality in the future. 
However, right now we are focusing on the V-SLAM aspect.

nominator, 2019-01-24:
@dorodnic That will be awesome. Until then, is it possible for 3rd party developers to use the raw streams of data from camera to implement these CV features on the host? Just want to make sure there is nothing to keep from it in the SDK or the device

dorodnic, 2019-01-24:
@nominator - the camera can provide:
1. Standalone 6-dof
2. Accel & Gyro
3. Dual fisheye monochrome (8-bit) streams @ 848x800 30FPS 

You can ask for any or all of these. 
When connected via USB3, you should be able to reliably get all together, when connected via USB2 we cannot guaranty no fisheye drops due to bandwidth.

We can publish additional sample data (*.bag) if you'd like to take a closer look at the streams (auto-exposure performance, IMU stability, etc...)

nominator, 2019-01-24:
@dorodnic That will be great!! Please do, so that until we get the camera shipped, we can start playing around with the data and think about how to best use this data to add more features required for AR applications. We are currently doing an AR MVP for a client and being able to show them advanced capabilities enabled by Realsense will be awesome.

mohammedari, 2019-01-24:
Cooool product I have waited for long time!

Does T265 have capabiliteis of realtime sharing of relocalization map between two devices over ethernet? 
I would like to put two T265 on my drones each and try to share their positions and orientations for colaborative task.

sirisian, 2019-01-25:
> Does T265 have capabiliteis of realtime sharing of relocalization map between two devices over ethernet?

I just pre-ordered one of these to play with and might buy more. This is exactly my question, and I think a lot of people will want/need this. Basically allow like 2+ sensors to work using the same shared internal map so they move in the same space. There are two use cases. One where two or more sensors are placed statically relative to each other and move together increasing their precision, and the other is where they move independently in the same space.

My high level plan is to track multiple controllers (and maybe HMDs?) turning them into inside-out trackers. I'd need an API feature that allows the sensors to synchronize/merge their internal maps wirelessly. (So this isn't a case where two sensors are attached to the same computer).

@dorodnic The image you linked shows 30 fps for the video. That's not the sampling rate is it?

dorodnic, 2019-01-26:
Hi @sirisian @mohammedari 
Sorry for the delay, I wanted to make sure to consult with my colleagues before giving an answer.
First, regarding the sampling rate - core pose data is available at 260 Hz. Fish-eye streams are at 30 FPS. 
Regarding multiple devices, what you are asking is not part of device's built-in capabilities, but rather a higher level use-case. The following should work - map the environment using one device, save the map to file, send the file over network and load it onto the second device. The second device should relocalize relative to the map, bringing the two devices into same coordinate space. 
Please note that the devices themselves do not have any connectivity on-board, nor the networking code will be provided in the SDK. It is too usage specific. 
Keeping devices synchronized is also up to you. There is no API (at least for the moment) to merge two maps, you can only fetch / store a map.

peci1, 2019-01-26:
@dorodnic If you could also create a bag file showing performance under the influence of vibrations, it'd be great. I mean vibrations from a drone, and vibrations from some kind of ground mobile robot/vehicle....

dorodnic, 2019-01-27:
Hi @peci1 
We will try to capture one, but I don't have something prepared. 
You are raising a good point. This is part of the reason why wheel odometry can be added as an additional input, and is recommended for robot scenarios.
Handheld and HMD-based use cases are easier in that regard, but we also did couple of test flights with the device, to make sure drone use-case can be supported. This is of course somewhat anecdotal evidence, so we will try to perform more evaluations and publish more data to help you make the right call.

delmottea, 2019-01-28:
the video stream is 30 fps, but is it also 30 fps used internally for the computation? I guess the accelerometer provides the 260Hz data and there is some interpolation between the processed images, but 30 fps seems a little bit low for fast movement (even though if it is global shutter, low exposure time and hardware sync, it may be enough).

What is the exposure time? is possible to modify it?

Is there a timestamp for the video stream that we can match to the position.

I saw it's not possible to add additional processing on top of the 6dof tracking, but is it possible to completely replacing the tracking by other task like would do on a normal movidius compute stick, to use it as a standard stereo camera with processing capability. I'm thinking of tracking or object detection that could be done on camera, and streaming only the detections instead of full image.

Is it possible to synchronize the clock of multiple cameras together like the d400 serie with a sync cable?

dorodnic, 2019-01-28:
Hi @delmottea -

> Is it possible to synchronize the clock of multiple cameras together like the d400 serie with a sync cable?

There is no external sync connector on this device. 

> What is the exposure time? is possible to modify it?

T265 performs auto-exposure on board. Actual exposure and gain are reported with every frame via metadata. 

If you are implementing your own 6-DOF algorithm (and not enabling the POSE stream) you can also set manual exposure value (this is not yet in LRS API, but should be possible according to device specifications) 

> the video stream is 30 fps, but is it also 30 fps used internally for the computation? I guess the accelerometer provides the 260Hz data and there is some interpolation between the processed images, but 30 fps seems a little bit low for fast movement
 
IMU rates are 200 Hz for Gyro and 62 Hz for Accel. Every 33ms the device corrects IMU drift using visual data from fisheye cameras. 

> Is there a timestamp for the video stream that we can match to the position.

Yes, there are hardware timestamps on all streams, and a method to query device current time (the later is not yet exposed in LRS, but will be)

> I saw it's not possible to add additional processing on top of the 6dof tracking, but is it possible to completely replacing the tracking by other task like would do on a normal movidius compute [...]

Unlike the NCS, the T265 is not a general purpose compute device, meaning we do not provide an SDK for development of custom firmware on top of its sensors, at least not at the moment. 

This device was designed and optimized for the problem of inside out tracking. Low-power plug&play V-SLAM is significant part of its value proposition. For general purpose CV an NCS + standard webcam  would probably make more sense.

Mrginbringer, 2019-02-04:
@dorodnic What type of wheel odometer would you recommend us using when feeding data fo the T265 camera?

dorodnic, 2019-02-05:
Hi @nominator - the response that I got on this question is to expect the map to hold roughly couple of rooms / house scale, but it depends on multiple factors such as how complicated the space is. To handle this, some kind of tiling mechanism can be established. The API we have for maps only allow to download / upload the entire map. Streaming map data might be implemented as a future feature request, but not available right now. 

@Mrginbringer - we have tested with the one that comes with Kabuki turtlebot, but it should work with any velocimeter/odometer.

dorodnic, 2019-02-05:
Got an Ok to redistribute the following video (link):
[image]
Running into problems getting the raw ros-bag, but will continue to publish the most information I can

Mrginbringer, 2019-02-05:
@dorodnic Whats the reason the T265 needs the wheel odometry? Have I gotten it completely wrong when visual odometry is far more exact when it comes to localization?

atyshka, 2019-02-05:
@Mrginbringer it's sensor fusion. Dead reckoning is subject to drift in the long term, but over a short distance wheel odometry provides fairly accurate readings to help establish the sensor's map. Slam mapping in general benefits from wheel odometry when visual odometry faces sudden changes.

dorodnic, 2019-02-05:
Hi @atyshka - well there are not that many features in this example. Device memory is limited (and I couldn't get a strict estimate on the memory size yet)
Stereo play very little part in tracking. Just to reiterate there is no depth provided from this device. Due to extreme FOV (that is beneficial for tracking) any depth it would produce would be pretty useless. 
There are two fish-eye cameras more for redundancy (less chance both will get blocked). The algorithm is fully functional with just one. 

@Mrginbringer - to add to @atyshka answer, we are aware that our implementation of SLAM is sensitive to vibrations, so in order to offer reliable and robust product, we recommend to add another data source. With wheel odometry the algorithm performs much nicer. Specifically, when there is a lot of noise in the IMU, the algorithm will rely too heavily on visual features, and wheel odometry help rectify sudden loss of visual features.

atyshka, 2019-02-05:
Alright thanks for the info. I'm not exactly sure what to expect from this device in terms of larger spaces, but I guess we'll just wait for the test units to ship to find out.

nominator, 2019-02-05:
@dorodnic thanks for explaining the map persistence. I hope map streaming makes it in the sdk soon because it will significantly improve the utility of the sensor in a lot of applications that require large scale mapping and relocalization.

filiperosa, 2019-02-06:
@dorodnic is it possible to feed it GPS position? Can it be used directly with a common flight controller such as Pixhawk?

dorodnic, 2019-02-06:
Hi @filiperosa 
As far as I know, at the moment we only provide API for wheel odometry input. 
I'm not sufficiently familiar with drones architecture to comment on integration with flight controllers, so you should assume the answer for now is no, at least for now. 
@nominator - thank you for the feedback, I'll make sure to relay it to the product team.

nominator, 2019-02-06:
@filiperosa You will need the Relasense SDK running on the flight controller to talk to T265, which I think is not possible with any flight controller on the market because these are microcontroller based hard realtime systems. 

You don't need to interface this sensor directly with a flight controller though, because the best way is to connect it to an on-board host computer to do compute heavy work and the host can talk to the FC over mavlink/ROS etc. you can get the 6dof measurements from T265 in the host and publish these as ROS position updates that pixhawk will happily consume in its own trajectory tracking process. While the host can use it for planning and navigation at the same time.

Can you explain why you need to interface directly with an FC?

dorodnic, 2019-02-06:
Longer T265 recording including several flights of stairs and a daylight outdoor section: 
[image]
[20190115_TM265.zip (**4GB compressed, 9GB bag**)](http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/20190115_TM265.zip)

atyshka, 2019-02-06:
@filiperosa No you cannot, the flight controller has neither usb nor the ability to run librealsense. You will need a host computer like an intel NUC or ODROID commanding the flight controller.

MartyG-RealSense, 2019-02-06:
DroneCode PX4 has compatibility with D435 when used with the Intel Aero Compute Board, according to its instructions.  

https://docs.px4.io/en/complete_vehicles/intel_aero.html?language=en_US

RealSense SDK 2.0 can be installed onto the Aero board, as described in the instructions linked to below headed 'Intel RealSense SDK - main branch (for D435)'.

https://github.com/intel-aero/meta-intel-aero/wiki/90-%28References%29-OS-user-Installation#intel-realsense-sdk---main-branch-for-d435

 So if the SDK supports T265 then you would expect in theory that it would be able to interact with a T265 attached to the Aero board.  In theory at least.  I could not give assurance about how well it works until somebody actually tries such a setup.

skohlbr, 2019-02-07:
Given the sensor can be configured to only stream pose data, will it be possible to operate it from a USB 2.0 port? It sounds that in principle, that should be possible as long as one is not interested in the image streams. Or has USB 2.0 compatibility been removed for the T265?

MartyG-RealSense, 2019-02-07:
I noticed that a guide to getting IMU data from D435i and T265 has been published.

https://realsense.intel.com/how-to-getting-imu-data-from-d435i-and-t265/

mikeh9, 2019-02-12:
Need some advice. We are needing the smallest physical size and power footprint for a host to interface with the T265. Only the pose data is needed so a USB2 interface should be sufficient.
It says the host is required to run Windows or Linux with approximately 40 MB persistent memory to store the camera firmware.
1. Can something as simple as a Raspberry Pi Zero work for the host or is there something smaller and less power hungry that would work?

MartyG-RealSense, 2019-02-12:
I would imagine that you should be able to connect a T265 to a Raspberry Pi Zero's USB OTG (On The Go) port via an OTG adapter.  Whilst RealSense cameras such as the D415 and D435 can sometimes have problems with power stability on OTG connections with single-board computing devices such as the Up Board and Intel Compute Stick, the T265 should be less affected by this possibility.

You can research alternative boards by googling for 'smallest single board computer', which gives results such as this:

http://linuxgizmos.com/worlds-smallest-quad-core-sbc-starts-at-8-dollars/

Jeremyye114, 2019-02-18:
Can T265 catch Skeletal animation???

dorodnic, 2019-02-18:
@Jeremyye114 - no, it's pretty much just the 6-dof tracking. You can however use it as a regular fisheye webcam and perform any CV you like on the host.

MartyG-RealSense, 2019-02-18:
I just wanted to highlight to everyone on this discussion that at the January 2019 Vision Industry and Technology Forum, Roy Karunakaran, the Director of Marketing for RealSense, demonstrated use of the D435i depth camera and the T265 tracking camera to operate a wheeled robot.

https://www.embedded-vision.com/industry-analysis/video-interviews-demos/intel-demonstration-realsense-cameras-visual-navigation-whe

adamkhalif, 2019-02-20:
Hello!

I'm working on a project where we are planning to use T265 for positioning of an autonomous car. Our camera is arriving in a few weeks hopfully, but  we are on a tight schedule and our goal is to have implemented the wheel odometry by then.

Exactly what input does the T265 need from the wheel odometry? Do we need one sensor or each wheel or is one or two enough? Does it require input in terms of position, speed, velocity or acceleration? Could it be possible to use the speed information from the cars integrated system or is it to inexact?

Is there any more detailed documentation available at the moment or do you know when it might be available?

BR,
Adam

MartyG-RealSense, 2019-02-21:
A data sheet document is available for the T265.

https://www.intel.com/content/www/us/en/support/articles/000032422/emerging-technologies/intel-realsense-technology.html

abylikhsanov, 2019-02-22:
I have a question regarding the vSLAM. What is the best way to perform the vSLAM using this camera? I thought to get the point cloud and then use the point cloud in ROS to construct a map but I am pretty sure I am wrong here. I want to place the camera on a mobile robot 2-3 cm high up from the floor

MartyG-RealSense, 2019-02-22:
At the January 2019 Vision Industry and Technology Forum, Roy Karunakaran, the Director of Marketing for RealSense, demonstrated use of the D435 and the T265 tracking camera to operate a wheeled robot.   I hope this will give some helpful insight about practical operation of the T265.

The T265 does not have depth sensing, so needs to be paired with another camera such as a D415 or D435 for applications where depth is required.  Dorodnic the RealSense SDK Manager quotes from the official FAQ at the top of this discussion:  

*********

* **Is this a Depth-Camera?**
No, the T265 does not offer any Depth information. It can provide IMU (Accel and Gyro), two monochrome fisheye streams, and 6-DOF pose data.

* **Can I use it together with a Depth-Camera?**
Yes, this device can be used with RealSense (and most likely any 3rd-party) depth cameras. It is passive and comes with an IR-cut filter.

**********

https://www.embedded-vision.com/industry-analysis/video-interviews-demos/intel-demonstration-realsense-cameras-visual-navigation-whe

abylikhsanov, 2019-02-22:
It does not really. The video did not show any mapping except for the odometry. So getting back to the question, what is the preferred way to do the SLAM? Is it by extracting features from the images and creating a PCL? The website states that it is intended for SLAM and it is a standalone camera, so I do not want to combine with any depth cameras and use only D245 for SLAM. I think I am asking a reasonable question here.

MartyG-RealSense, 2019-02-22:
Doing SLAM with just a D435i (the D435 depth camera model with an IMU) is certainly very viable, as described in the link below.  Reading this page will explain the process in this SLAM solution that generates the mapping and localization.

https://github.com/intel-ros/realsense/wiki/SLAM-with-D435i?language=en_US

abylikhsanov, 2019-02-22:
Thank you but I was asking about D245 and not about the depth camera :)

MartyG-RealSense, 2019-02-22:
I understand.   I meant that a single D435i camera is an option for SLAM if you did not want to pair a depth camera and the T265 tracking camera together.  

In regard to T265, any application that requires depth data is going to need the T265 to be paired with a depth camera, as the T265 alone cannot perform that depth-sensing operation.  The sensors that the T265 does have is dual fisheye cameras and an IMU.

atyshka, 2019-02-22:
@abylikhsanov I understand what you’re saying about it being branded as a slam sensor without depth data. I assume you’re talking about the t265, there is no realsense d245. Whereas normal slam like lidar or rtab-map create a point cloud map of features, this sensor does the mapping with visual features. Both this map and a rough occupancy map are accessible from the T265. This comprises the “mapping” part of slam. If you want a precision pointcloud, as @MartyG-RealSense commented, you will have to use a t265 with a depth camera or just a standalone depth camera.

nominator, 2019-02-22:
@abylikhsanov To further build upon what @atyshka and @MartyG-RealSense explained above. 

T265 is all about tracking and mapping. With just T265 you should be able to move around a get your position and a map of the surrounding (map is based on features detected visually). I not aware of the actual data format of the map but for the sake of completeness, it will be safe to say that it will constitute the following

1- An origin (usually the position of device where the mapping session started)
2- All detected features represented internally in a data format that describes the properties of the features and their positions w.r.t. the origin (this also roughly defines the extents/bounds of the map implicitly)
3- At runtime a current 6DOF position of the device. 

When you load an existing map, the device collects the current frame from camera and compares its features with the ones saved in the map and if it finds enough matches your device gets relocalized within that map.

If you have other use cases where you need to detect collisions with space geometry (occupancy map for navigation, virtual and real object occlusion, procedural content generation etc.) then you need a dense depth map of the environment. For this you will need to use any D4XX camera. 

Depending on application you may only require depth or tracking or both. To answer your question about using pointcloud in RTabMap to calculate pose, with T265 there is no need, everything is done onboard camera module and you get the end result sent to you, a 6DOF pose. If you want to calculate a pointcloud using T265, then I think it is not exposed in the API, I could be wrong, but even if it were, it will be a sparse point cloud which may not be of any use apart from mapping and tracking. For all proper depth related use cases you need a depth camera.

MartyG-RealSense, 2019-02-22:
@nominator Thanks so much for your explanation!

if the term 'realsense t265 slam' is googled for, then many search results of news articles about the T265 give the impression that SLAM is something that the T265 can do on its own unless you read further into the articles and discover that the T265 complements the 400 Series rather than replaces it.

abylikhsanov, 2019-02-22:
@nominator @atyshka 

Thank you for the detailed explanations. So basically T265 is mainly used to localize itself in the existing map because depth cameras like D435 will perform actual mapping (may be not SLAM) better as it gets dense PCL.

But on the other hand, I think algorithms like orb slam can perform a very good localization on an existing map having D435 with global shutter. So I am confused which camera to purchase. Right now it seems that D435 is the camera that I need mainly because I have an indoor robot that needs to perform SLAM and navigate afterwards and avoid the obstacles in front of it.

nominator, 2019-02-22:
@abylikhsanov  No, T265 creates its own map and localizes in it using SLAM. 

D435 is used to create depth maps (point clouds with each pixel having a depth value in addition to an x,y position in the frame). Depth map is refreshed every frame there is no relation between depth maps of 2 different frames. Please don't confuse a depth map with a SLAM based map, a SLAM based map is built over time by fusing sensor measurements from IMU and the fisheye cameras and can be used for estimating position and tracking it.

Please read about SLAM and stereo depth. There are lot of articles and tutorials available online. Without a basic understanding of these techniques/algorithms you will have a hard time understanding the purpose and function of these devices (T265 and D435).

nominator, 2019-02-22:
@abylikhsanov Either you can use an iD435 with a SLAM library like Orbslam or Rtabmap to do navigation/tracking and collision avoidance both. However the navigation quality may not be as good since the limited FOV of the RGB camera may not be able to find enough features to estimate an accurate pose.

Ideal solution will be to use both a T265 and D435. Where T265 will perform mapping and localization while the depth map can be used to do collision avoidance.

abylikhsanov, 2019-02-22:
@nominator I do not think that this is due to the lack of knowledge in SLAM. I never told that depth map is SLAM. What I said is dense PCL can be easily converted to other numerical values which are essential for SLAM.

nominator, 2019-02-22:
SLAM works with identifying features in a frame and then searching and matching the same features from a different camera angle and position. The difference in appearance of same features from different frames is used to estimate the position of the camera. These features are points in a frame that have certain properties. These properties are important as it helps to search them in other frames. This is an oversimplified explanation but is basically what happens.

Even if you have a dense depth frame, that does not warrant that you will be able to compare and match  points in this depth frame with points from another depth frame. This is required to estimate pose of the camera.

What you need is a good amount of feature points that have more chance of being matched with data from another frame. Wider FOV cameras help in this. D435 has wide FOV IR cameras but then there is IR pattern being projected on the scene which can confuse the feature identification. So it is possible to do SLAM with D435 but not ideal.

abylikhsanov, 2019-02-22:
@nominator Again, from your writings you mention that T265 should perform SLAM which I honestly disagree after mentioning the rtabmap as an example. Also from your last message you keep mentioning SLAM bu talk about localization. 
Localization does not need to be feature based but mapping shuld. So from the simple logic, if using Rtabmap, which requires an accurate dense data, RGB-D camera is perferred in indoor scenario as there is no "guessing" like stereo camera would do.

nominator, 2019-02-22:
SLAM = simultaneous localization and mapping and both mapping and localization inform each other to improve the estimate over time. 

D435 calculated depth based on stereo matching.

abylikhsanov, 2019-02-22:
I have accidentally deleted my previous comment, I wanted to edit it. Basically, I just meant that I disagree with your opinion that stereo should be used as SLAM and you are mostly talking about localization

abylikhsanov, 2019-02-22:
Okay, I don't think that we should continue this argument as it is not a private message.

atyshka, 2019-02-22:
Honestly I too believe the SLAM label is a bit of a misnomer. Yes it localizes, but the mapping isn't very useful. Wheres with traditional slam I can make an occupancy map to avoid collision, I can't do that with this sensor. The map is basically a bunch of "keypoints" the provide anchors for localization but those keypoints are hardly a map of the environment. In order to truly map in a useful way, a combination of T265 and D435 would be needed. That is how I plan to use the sensor.

manomitbal, 2019-02-22:
Hi!
Thank you for the detailed assessment of the device! In my experience using visual feature based SLAM or pose estimation algorithms with the cameras, I have always encountered scenarios where tracking fails whenever the device loses features due to sparse information in the environment (like a featureless wall). Is there a way the T265 overcomes this issue? I believe that having a tightly coupled hardware synced IMU would surely assist in retaining tracking information when visual features are lost but the I am uncertain about the noise metrics in the IMU on board to solely support tracking until the device encounters visual features again. As @dorodnic mentioned before, that the addition of wheel encoder information greatly improves the algorithm performance, what exactly is the modality of input for that? Is it something like a loosely coupled EKF combining pose information from the camera and the wheel encoder data or is the wheel encoder data tightly coupled by pushing the data as input to the device directly (as it says on the website -- the wheel odometry inputs can be piped directly to the device over USB).

Thank you.

[93 more replies]

closed 2019-04-04 by RealSenseCustomerSupport