realsenseai/librealsense #3131: Install vs. build from the source on Windows

YangJiao1996, 2019-01-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Windows 10 1609 | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.3                         | 
| Language                            |  C++                      | 

### Issue Description
I recently installed RealSense SDK on Windows and tried to develop some apps based on the C++ examples. Are there any differences between installing the SDK using the Intel.RealSense.SDK.exe and building the SDK from source?   

I asked this question because I am having troubles in compiling the examples manually using CMake and Visual Studio. It looks like the RealSense libraries cannot be automatically found by the CMakelists.txt files provided in the example folders. I just wonder whether it requires the SDK to be built from source for CMake to work.

dorodnic, 2019-01-23:
Hi @YangJiao1996 
There is no difference in source code between the two. With the Windows installer we also provide some manually configured project files, but if you want to use CMake, you need to configure the CMakeLists.txt yourself. We have such example for [Linux](https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake), but not yet for Windows.

YangJiao1996, 2019-01-23:
Thank you @dorodnic

closed 2019-01-23 by YangJiao1996