realsenseai/librealsense #3135: Multi camera calibration

EmmanuelCos, 2019-01-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Win 10  | 
| Language                            |  C++       | 
| Segment			|  object reconstruction        | 

Hello, 
I am using four cameras in parallel, at different constant position, so that I can capture a 360° 3D picture of any random object.
To link each image taken by each camera, I only performed rotations and translations, knowing a priori the theoretical positions of each camera. The fact is that I have error measures and there is a little gap between each image when I do the reconstruction. 
Do you know how I could do a dynamic calibration? I could for instance put a visible point in the field of vision of the four cameras, detect this point, and then do the rotations and translation according to the distances from this point, and not according to the theoretical positions of the cameras i have?
Has anyone already done that kind of things, or has a good method to perform multicamera calibration?

Thank you and regards,

Emmanuel

MartyG-RealSense, 2019-01-24:
A tool called Vicalib, which Intel themselves use, can be used for multi-camera calibration.  It uses a board that you show to each of the cameras in turn..  Below is a link to the Vicablib software.

https://github.com/arpg/vicalib?language=en_US

EmmanuelCos, 2019-01-24:
Thank you MartyG! 
The documentation for the installation is only for UNIX and Mac. Do you know if it's actually possible to install this library on windows 10, and if there is an existing documentation for that?
Thank you,
Emmanuel

MartyG-RealSense, 2019-01-24:
I would imagine that it is possible to compile the source code for Windows 10, and that the documentation chooses just to focus on compilation methods for Linux and Mac.  

The documentation mentions that Vicalib can be built with the GCC compiler.  A Windows variant of GCC is MinGW.

https://sourceforge.net/projects/tdm-gcc/

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

@EmmanuelCos I'm afraid that there's no document for Vicalib installation on windows yet. Maybe you can have a try on Linux or Mac first.

EmmanuelCos, 2019-02-12:
I think that intel should develop one user friendly tool for multi camera calibration. 
In librealsense there is already a multicam part, and I appreciate the effort, but it is of little use if there is nothing for the calibration. 
Do you know to whom I should ask for that?

MartyG-RealSense, 2019-02-12:
If Intel add a feature request tag to this discussion then it will make it easier for them to track it over time.

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

There are also some good examples on ROS section to demonstrate using multi-camera that are installed at fixed locations.
https://github.com/intel-ros/realsense/wiki
Examples can be found on the right hand side of the page.

closed 2019-03-18 by RealSenseCustomerSupport