realsenseai/librealsense #3136: Syncronized recordingin the unity wrapper

Dealerpriest, 2019-01-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | depends on which version the unity package uses. 2.18.0.515 installed with win installer. But the wrapper seems to be based on v2.16.0? |
| Unity wrapper | unity wrapper installed with the downloadable unity package from [here](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.0/Intel.RealSense.unitypackage) |
| Language                            |  unity / C#                          | 
| Segment			|                          | 

### Issue Description
When connecting several cameras (d415) and enable recording, the cameras won't start recording at the same time. The record is started with offsets in time and the recorded bag-files are thus unsynchronized. How can I achieve syncronized record (and playback) in unity, using several d415 depth cameras?

I had an idea that the sync problem could be tackled by pausing the recording right after `cfg.EnableRecordToFile(RecordPath);` in RsConfiguration (or somewhere close). Then resume the recordings as soon as I have triggered recordmode for all the cameras. Would that work?

I found the `RecordDevice.Pause()` api and thought that would be perfect. But I can't find a way to instantiate/cast the device as a RecordDevice. I could I achieve that?

MartyG-RealSense, 2019-01-24:
RealSense community member @frenchtouchXR worked out a technique for recording multiple streams at the same time with multiple cameras by making a short edit to one of the RealSense Unity wrapper's files.

https://github.com/IntelRealSense/librealsense/issues/2203#issuecomment-412218681

ogoshen, 2019-01-28:
`RecordDevice` creation answered in #3142

Regrading this issue, I think you've got the right idea, but don't expect your multiple recordings to be fully synchronized.

First there's the [HW sync](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=hw+sync).
Then you'll need to make sure the frames are synced when recorded.
This could hopefully be solved using `RecordDevice.Pause\Resume`  but might be trickier than that...

You can maybe take a look at the `cs-tutorial-4-software-dev` example, the idea is that you can create a `SoftwareDevice` that feeds on frames from multiple cameras, and wrap it in a recorder.
It also uses a `Syncer` processing block, which tries to match frames by timestamps.

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Hi Dealerpriest,

Wonder if you have tried suggestions from MartyG and ogoshen. Any updates?

Thanks!

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi Dealerpriest,

Wonder if you still need help on this one?

Thanks!

Dealerpriest, 2019-02-14:
Thanks for the reply. I've unfortunately been very occupied in other projects. But I will be able to take a look tomorrow. I'll report back here then.

Dealerpriest, 2019-02-15:
For my purposes, it's not critical that the recordings are syncronized exactly to the frame. Rather that they are aligned well enough to look visually good. My conclusion was that this would be able to achieve if the recording (or the playback) is aligned in software. Do you think this is feasible?

ogoshen, 2019-02-17:
Sounds possible, as long as your scenes are relatively static SW sync should look good enough.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi Dealerpriest,

Wonder if you get any updates based on above discussions? Still need any help?

Thanks!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi Dealerpriest,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-03-21 by RealSenseCustomerSupport