realsenseai/librealsense #3137: Invalid Value in rs2_get_option error after upgrading D435i to firmware 5.11.1.0

ayu135, 2019-01-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 5.11.1.0 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0-39-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.18.0                          | 
| Language                            |  C++/ROS                          | 
| Segment			|  VR/AR                          | 

### Issue Description
Hi after I upgraded the firmware on my realsense D435i to 5.11.1.0 the camera no longer streams any data even on the realsense viewer. I get an error saying:
```
Invalid Value in rs2_get_option(options:0x21ce000, option:Emitter On Off):
hwmon command 0x7d failed. Error type: Invalid Command (-1).
```
I get the same error when running the rs_camera.launch file from the ROS wrapper and also get the same error in realsense-viewer. I tried various things like re-plugging the sensor, rebooting my PC and hardware resetting, nothing seems to solve this issue. I was able to get the realsense to work again but only if I downgrade the firmware to the previous version(5.10.13). I thing the error is specific to the latest 5.11 firmware.

ayu135, 2019-01-24:
You don't need to downgrade the librealsense SDK just the firmware of your device using the DFU tool

ev-mp, 2019-01-24:
Hello @ayu135 , 
The issue will be resolved by the following [PR](https://github.com/IntelRealSense/librealsense/pull/3127).
In the meanwhile you can either apply it manually,  or disregard.

ayu135, 2019-01-24:
Thanks, that solves it for me I am closing the issue

closed 2019-01-24 by ayu135

bormy, 2019-02-01:
@ayu135 would you mind sharing the step to resolve this issue?

Regards,
Bormy

dorodnic, 2019-02-10:
@bormy - if you download the latest version ([v2.18.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1)) you should no longer see the issue