realsenseai/librealsense #3138: Constructor rs2::context freezes

bansan85, 2019-01-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Windows 10 v 1803  | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.0 | 
| Language                            |  C     | 

### Issue Description
I build a C++ library that is linked to realsense.dll (64 bit).

I have a global variable : 
`rs2::context Camera::ctx;`

When I do a LoadLibrary of my dll, LoadLibrary freezes because of `rs2::context::context`. It's not always but it's seems to be the linker "fault" and not at runtime because it worked successfully for a month and since yesterday it's always freezing. I "think" it because the realsense library didn't finish to do all it's work and I call too early the context constructor.

To fix it, I convert the ctx to a pointer that I create dynamically.

Is that a bug or something else ?

Thanks,

Stacktrace (my dll is RealSenseHiris.dll) :
```
Not Flagged		9192	0	Main Thread	Main Thread	combase.dll!CRpcResolver::ServerRegisterOXID	Normal
 	 	 	 	 	 	ntdll.dll!NtAlpcSendWaitReceivePort()	 
 	 	 	 	 	 	rpcrt4.dll!LRPC_BASE_CCALL::SendReceive()	 
 	 	 	 	 	 	rpcrt4.dll!NdrpClientCall2()	 
 	 	 	 	 	 	rpcrt4.dll!NdrClientCall2()	 
 	 	 	 	 	 	combase.dll!CRpcResolver::ServerRegisterOXID(const __MIDL_ILocalObjectExporter_0007 & oxidInfo, unsigned __int64 * poxid, unsigned long * pcOidsToAllocate, unsigned __int64 * arNewOidList) Line 1135	 
 	 	 	 	 	 	combase.dll!OXIDEntry::RegisterOXIDAndOIDs(unsigned long * pcOids, unsigned __int64 * pOids) Line 1617	 
 	 	 	 	 	 	combase.dll!CComApartment::CallTheResolver() Line 863	 
 	 	 	 	 	 	combase.dll!CComApartment::InitRemoting() Line 1142	 
 	 	 	 	 	 	combase.dll!InitChannelIfNecessary() Line 1576	 
 	 	 	 	 	 	combase.dll!CGIPTable::RegisterInterfaceInGlobalHlp(IUnknown * pUnk, const _GUID & riid, unsigned long mshlflags, unsigned long * pdwCookie) Line 809	 
 	 	 	 	 	 	combase.dll!CGIPTable::RegisterInterfaceInGlobal(IUnknown * pUnk, const _GUID & riid, unsigned long * pdwCookie) Line 770	 
 	 	 	 	 	 	PortableDeviceApi.dll!ATL::CComGITPtr<struct IWpdSerializer>::operator=(struct IWpdSerializer *)	 
 	 	 	 	 	 	PortableDeviceApi.dll!CPortableDevice::Open(unsigned short const *,struct IPortableDeviceValues *)	 
 	 	 	 	 	 	SensorsApi.dll!DeviceOpen(unsigned short const *,struct IPortableDevice * *,int)	 
 	 	 	 	 	 	SensorsApi.dll!CSensor::Initialize(unsigned short const *,unsigned short const *,struct _GUID const &,struct _GUID const &,struct _GUID const &,class std::shared_ptr<class CClassExtensionProxy>,int)	 
 	 	 	 	 	 	SensorsApi.dll!CSensorManager::CreateV1Sensor(struct _DEV_OBJECT const *,struct ISensor * *)	 
 	 	 	 	 	 	SensorsApi.dll!CSensorManager::GetSensorsFromFilterExpression(struct _DEVPROP_FILTER_EXPRESSION const *,unsigned long,struct ISensorCollection * *)	 
 	 	 	 	 	 	SensorsApi.dll!CSensorManager::GetSensorsByInterface(struct _GUID const &,struct ISensorCollection * *)	 
 	 	 	 	 	 	SensorsApi.dll!CSensorManager::GetSensorsByCategory(struct _GUID const &,struct ISensorCollection * *)	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f5eb9f1()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f5f34f0()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f5ee686()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f5edfa0()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f5f1d0c()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f627a1d()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f680635()	 
 	 	 	 	 	 	realsense2.dll!00007ffa8f6963bf()	 
 	 	 	 	 	 	RealSenseHiris.dll!rs2::context::context() Line 101	 
 	 	 	 	 	 	RealSenseHiris.dll!`dynamic initializer for 'Camera::ctx''() Line 9	 
 	 	 	 	 	 	ucrtbased.dll!_initterm(void(*)() * first, void(*)() * last) Line 22	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_crt_process_attach(HINSTANCE__ * const instance, void * const reserved) Line 67	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_crt_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 133	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 190	 
 	 	 	 	 	 	RealSenseHiris.dll!_DllMainCRTStartup(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 249	 
 	 	 	 	 	 	ntdll.dll!LdrpCallInitRoutine()	 
 	 	 	 	 	 	ntdll.dll!LdrpInitializeNode()	 
 	 	 	 	 	 	ntdll.dll!LdrpInitializeGraphRecurse()	 
 	 	 	 	 	 	ntdll.dll!LdrpPrepareModuleForExecution()	 
 	 	 	 	 	 	ntdll.dll!LdrpLoadDllInternal()	 
 	 	 	 	 	 	ntdll.dll!LdrpLoadDll()	 
 	 	 	 	 	 	ntdll.dll!LdrLoadDll()	 
 	 	 	 	 	 	KernelBase.dll!LoadLibraryExW()	 
 	 	 	 	 	 	mfc140ud.dll!IsolationAwareLoadLibraryW(const wchar_t * lpLibFileName) Line 136	 
 	 	 	 	 	 	mfc140ud.dll!AfxLoadLibrary(const wchar_t * lpszModuleName) Line 182	 
 	 	 	 	 	 	MFCApplication1.exe!CMFCApplication1App::InitInstance() Line 76	 
 	 	 	 	 	 	mfc140ud.dll!AfxWinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, wchar_t * lpCmdLine, int nCmdShow) Line 37	 
 	 	 	 	 	 	MFCApplication1.exe!wWinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, wchar_t * lpCmdLine, int nCmdShow) Line 26	 
 	 	 	 	 	 	MFCApplication1.exe!invoke_main() Line 118	 
 	 	 	 	 	 	MFCApplication1.exe!__scrt_common_main_seh() Line 253	 
 	 	 	 	 	 	MFCApplication1.exe!__scrt_common_main() Line 296	 
 	 	 	 	 	 	MFCApplication1.exe!wWinMainCRTStartup() Line 17	 
 	 	 	 	 	 	kernel32.dll!BaseThreadInitThunk()	 
 	 	 	 	 	 	ntdll.dll!RtlUserThreadStart()	 
```

dorodnic, 2019-01-24:
Hi @bansan85 
I'd recommend wrapping the thing like this:
```cpp
rs2::context& global_context()
{
     static rs2::context ctx;
     return ctx;
}
```
This way it is allocated on first request, and is also thread-safe.

bansan85, 2019-01-25:
Thanks, it's better than a `new`.

I still leave the issue open because I think that `rs2::context::context()` shouldn't be able to hang an application. I would rather an exception.

dorodnic, 2019-01-27:
Sure, that's reasonable. It will automatically get escalated to Intel support and make its way to our internal backlog, but can take a while before addressed, since this is a fairly niche use-case.
Also, the call-stack concerns me - since the dead-lock originates outside librealsense, it most likely has something to do with COM initialisation rules and quirks, and can be pretty hard to root-cause.

bansan85, 2019-01-28:
I would like to update the stack with debug symbol when I use global variable for `rs2::context`.
```
Not Flagged	>	14096	0	Main Thread	Main Thread	combase.dll!CRpcResolver::ServerRegisterOXID	Normal
 	 	 	 	 	 	ntdll.dll!NtAlpcSendWaitReceivePort()	 
 	 	 	 	 	 	rpcrt4.dll!LRPC_BASE_CCALL::SendReceive()	 
 	 	 	 	 	 	rpcrt4.dll!NdrpClientCall2()	 
 	 	 	 	 	 	rpcrt4.dll!NdrClientCall2()	 
 	 	 	 	 	 	combase.dll!CRpcResolver::ServerRegisterOXID(const __MIDL_ILocalObjectExporter_0007 & oxidInfo, unsigned __int64 * poxid, unsigned long * pcOidsToAllocate, unsigned __int64 * arNewOidList) Line 1135	 
 	 	 	 	 	 	combase.dll!OXIDEntry::RegisterOXIDAndOIDs(unsigned long * pcOids, unsigned __int64 * pOids) Line 1617	 
 	 	 	 	 	 	combase.dll!CComApartment::CallTheResolver() Line 863	 
 	 	 	 	 	 	combase.dll!CComApartment::InitRemoting() Line 1142	 
 	 	 	 	 	 	combase.dll!InitChannelIfNecessary() Line 1576	 
 	 	 	 	 	 	combase.dll!CDllHost::GetApartmentActivators(unsigned long & hCOMActivator, unsigned long & hWinRTActivator) Line 513	 
 	 	 	 	 	 	combase.dll!CClassCache::GetActivatorFromDllHost(int fSixteenBit, int isActivatingAppInstalledClass, unsigned long dwDllThreadModel, unsigned long * phActivator) Line 3669	 
 	 	 	 	 	 	combase.dll!CProcessActivator::GetApartmentActivator(ActivationPropertiesIn * pInActProperties, ISystemActivator * * ppActivator) Line 1093	 
 	 	 	 	 	 	combase.dll!CProcessActivator::CCICallback(unsigned long dwContext, IUnknown * pUnkOuter, ActivationPropertiesIn * pActIn, IActivationPropertiesIn * pInActProperties, IActivationPropertiesOut * * ppOutActProperties) Line 1601	 
 	 	 	 	 	 	combase.dll!CProcessActivator::AttemptActivation(ActivationPropertiesIn * pActIn, IUnknown * pUnkOuter, IActivationPropertiesIn * pInActProperties, IActivationPropertiesOut * * ppOutActProperties, HRESULT(CProcessActivator::*)(unsigned long, IUnknown *, ActivationPropertiesIn *, IActivationPropertiesIn *, IActivationPropertiesOut * *) pfnCtxActCallback, unsigned long dwContext) Line 1524	 
 	 	 	 	 	 	combase.dll!CProcessActivator::ActivateByContext(ActivationPropertiesIn * pActIn, IUnknown * pUnkOuter, IActivationPropertiesIn * pInActProperties, IActivationPropertiesOut * * ppOutActProperties, HRESULT(CProcessActivator::*)(unsigned long, IUnknown *, ActivationPropertiesIn *, IActivationPropertiesIn *, IActivationPropertiesOut * *) pfnCtxActCallback) Line 1368	 
 	 	 	 	 	 	combase.dll!CProcessActivator::CreateInstance(IUnknown * pUnkOuter, IActivationPropertiesIn * pInActProperties, IActivationPropertiesOut * * ppOutActProperties) Line 1268	 
 	 	 	 	 	 	combase.dll!ActivationPropertiesIn::DelegateCreateInstance(IUnknown * pUnkOuter, IActivationPropertiesOut * * ppActPropsOut) Line 1906	 
 	 	 	 	 	 	combase.dll!CClientContextActivator::CreateInstance(IUnknown * pUnkOuter, IActivationPropertiesIn * pInActProperties, IActivationPropertiesOut * * ppOutActProperties) Line 567	 
 	 	 	 	 	 	combase.dll!ActivationPropertiesIn::DelegateCreateInstance(IUnknown * pUnkOuter, IActivationPropertiesOut * * ppActPropsOut) Line 1958	 
 	 	 	 	 	 	combase.dll!ICoCreateInstanceEx(const _GUID & OriginalClsid, IUnknown * punkOuter, unsigned long dwClsCtx, _COSERVERINFO * pServerInfo, unsigned long dwCount, unsigned long dwActvFlags, tagMULTI_QI * pResults, ActivationPropertiesIn * pActIn) Line 1959	 
 	 	 	 	 	 	combase.dll!CComActivator::DoCreateInstance(const _GUID & Clsid, IUnknown * punkOuter, unsigned long dwClsCtx, _COSERVERINFO * pServerInfo, unsigned long dwCount, tagMULTI_QI * pResults, ActivationPropertiesIn * pActIn) Line 385	 
 	 	 	 	 	 	combase.dll!CoCreateInstance(const _GUID & rclsid, IUnknown * pUnkOuter, unsigned long dwContext, const _GUID & riid, void * * ppv) Line 121	 
 	 	 	 	 	 	realsense2.dll!librealsense::platform::wmf_hid_device::foreach_hid_device(std::function<void __cdecl(librealsense::platform::hid_device_info,ATL::CComPtr<ISensor>)> action) Line 363	 
 	 	 	 	 	 	realsense2.dll!librealsense::platform::wmf_backend::query_hid_devices() Line 134	 
 	 	 	 	 	 	realsense2.dll!librealsense::platform::win_event_device_watcher::win_event_device_watcher(const librealsense::platform::backend * backend) Line 149	 
 	 	 	 	 	 	realsense2.dll!std::make_shared<librealsense::platform::win_event_device_watcher,librealsense::platform::wmf_backend const * __ptr64 const>(const librealsense::platform::wmf_backend * const && <_Args_0>) Line 971	 
 	 	 	 	 	 	realsense2.dll!librealsense::platform::wmf_backend::create_device_watcher() Line 344	 
 	 	 	 	 	 	realsense2.dll!librealsense::context::context(librealsense::backend_type type, const char * filename, const char * section, rs2_recording_mode mode, std::basic_string<char,std::char_traits<char>,std::allocator<char> > min_api_version) Line 132	 
 	 	 	 	 	 	realsense2.dll!std::make_shared<librealsense::context,enum librealsense::backend_type>(librealsense::backend_type && <_Args_0>) Line 971	 
 	 	 	 	 	 	realsense2.dll!rs2_create_context(int api_version, rs2_error * * error) Line 136	 
 	 	 	 	 	 	RealSenseHiris.dll!rs2::context::context() Line 101	 
 	 	 	 	 	 	RealSenseHiris.dll!`dynamic initializer for 'ctx''() Line 13	 
 	 	 	 	 	 	ucrtbased.dll!_initterm(void(*)() * first, void(*)() * last) Line 22	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_crt_process_attach(HINSTANCE__ * const instance, void * const reserved) Line 67	 
[... cut ...]
 	 	 	 	 	 	mfc100u.dll!AfxWndProcBase(struct HWND__ *,unsigned int,unsigned __int64,__int64)	 
 	 	 	 	 	 	user32.dll!UserCallWinProcCheckWow()	 
 	 	 	 	 	 	user32.dll!DispatchMessageWorker()	 
 	 	 	 	 	 	mfc100u.dll!AfxInternalPumpMessage(void)	 
 	 	 	 	 	 	mfc100u.dll!CWinThread::Run(void)	 
 	 	 	 	 	 	mfc100u.dll!AfxWinMain(struct HINSTANCE__ *,struct HINSTANCE__ *,wchar_t *,int)	 
 	 	 	 	 	 	Hiris.exe!__tmainCRTStartup() Line 547	 
 	 	 	 	 	 	kernel32.dll!BaseThreadInitThunk()	 
 	 	 	 	 	 	ntdll.dll!RtlUserThreadStart()	 

```

bansan85, 2019-01-28:
I noticed a problem when I want to unload the library. The stack is different when I use `rs2::context` with your method `global_context()` :
```
Not Flagged	>	14120	0	Main Thread	Main Thread	realsense2.dll!Concurrency::details::stl_condition_variable_win7::wait	Normal
 	 	 	 	 	 	ntdll.dll!NtWaitForAlertByThreadId()	 
 	 	 	 	 	 	ntdll.dll!RtlSleepConditionVariableSRW()	 
 	 	 	 	 	 	KernelBase.dll!SleepConditionVariableSRW()	 
 	 	 	 	 	 	realsense2.dll!Concurrency::details::stl_condition_variable_win7::wait(Concurrency::details::stl_critical_section_interface * lock) Line 210	 
 	 	 	 	 	 	realsense2.dll!do_wait(_Cnd_internal_imp_t * cond, _Mtx_internal_imp_t * mtx, const xtime * target) Line 77	 
 	 	 	 	 	 	realsense2.dll!perc::Dispatcher::sendMessage<perc::Message>(perc::EventHandler * handler, const perc::Message & msg, int priority) Line 237	 
 	 	 	 	 	 	realsense2.dll!perc::Manager::processMessage(const perc::Message & msg, int prio) Line 218	 
 	 	 	 	 	 	realsense2.dll!perc::Manager::done() Line 164	 
 	 	 	 	 	 	realsense2.dll!perc::Manager::~Manager() Line 92	 
 	 	 	 	 	 	realsense2.dll!perc::Manager::`scalar deleting destructor'(unsigned int)	 
 	 	 	 	 	 	realsense2.dll!perc::TrackingManager::ReleaseInstance(perc::TrackingManager * & manager) Line 58	 
 	 	 	 	 	 	realsense2.dll!std::_Ref_count_del<perc::TrackingManager,<lambda_1894e2292d327cca7312a33063adecd3> >::_Destroy() Line 186	 
 	 	 	 	 	 	realsense2.dll!librealsense::tm2_context::~tm2_context() Line 60	 
 	 	 	 	 	 	realsense2.dll!librealsense::context::~context() Line 308	 
 	 	 	 	 	 	realsense2.dll!rs2_context::~rs2_context() Line 70	 
 	 	 	 	 	 	realsense2.dll!rs2_delete_context(rs2_context * context) Line 142	 
 	 	 	 	 	 	RealSenseHiris.dll!std::_Ref_count_del<rs2_context,void (__cdecl*)(rs2_context * __ptr64)>::_Destroy() Line 186	 
 	 	 	 	 	 	RealSenseHiris.dll!std::_Ref_count_base::_Decref() Line 113	 
 	 	 	 	 	 	RealSenseHiris.dll!std::_Ptr_base<rs2_context>::_Decref() Line 339	 
 	 	 	 	 	 	RealSenseHiris.dll!std::shared_ptr<rs2_context>::~shared_ptr<rs2_context>() Line 567	 
 	 	 	 	 	 	RealSenseHiris.dll!rs2::context::~context()	 
 	 	 	 	 	 	RealSenseHiris.dll!`GetCtx'::`2'::`dynamic atexit destructor for 'ctx''()	 
 	 	 	 	 	 	ucrtbased.dll!_execute_onexit_table::__l2::<lambda>() Line 206	 
 	 	 	 	 	 	ucrtbased.dll!__crt_seh_guarded_call<int>::operator()<void <lambda>(void),int <lambda>(void) & __ptr64,void <lambda>(void) >(__acrt_lock_and_call::__l2::void <lambda>(void) && setup, _execute_onexit_table::__l2::int <lambda>(void) & action, __acrt_lock_and_call::__l2::void <lambda>(void) && cleanup) Line 204	 
 	 	 	 	 	 	ucrtbased.dll!__acrt_lock_and_call<int <lambda>(void) >(const __acrt_lock_id lock_id, _execute_onexit_table::__l2::int <lambda>(void) && action) Line 912	 
 	 	 	 	 	 	ucrtbased.dll!_execute_onexit_table(_onexit_table_t * table) Line 231	 
 	 	 	 	 	 	RealSenseHiris.dll!__scrt_dllmain_uninitialize_c() Line 397	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_crt_process_detach(const bool is_terminating) Line 109	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_crt_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 134	 
 	 	 	 	 	 	RealSenseHiris.dll!dllmain_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 207	 
 	 	 	 	 	 	RealSenseHiris.dll!_DllMainCRTStartup(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 249	 
 	 	 	 	 	 	ntdll.dll!LdrpCallInitRoutine()	 
 	 	 	 	 	 	ntdll.dll!LdrShutdownProcess()	 
 	 	 	 	 	 	ntdll.dll!RtlExitUserProcess()	 
 	 	 	 	 	 	kernel32.dll!ExitProcessImplementation()	 
 	 	 	 	 	 	msvcr100.dll!doexit(int code, int quick, int retcaller) Line 621	 
 	 	 	 	 	 	Hiris.exe!__tmainCRTStartup() Line 568	 
 	 	 	 	 	 	kernel32.dll!BaseThreadInitThunk()	 
 	 	 	 	 	 	ntdll.dll!RtlUserThreadStart()	 
```

and when I use rs2::context as local variable (I noticed I use it only once) :

```
Not Flagged	>	14348	0	Main Thread	Main Thread	realsense2.dll!Concurrency::details::stl_condition_variable_win7::wait_for	Normal
 	 	 	 	 	 	ntdll.dll!NtWaitForAlertByThreadId()	 
 	 	 	 	 	 	ntdll.dll!RtlSleepConditionVariableSRW()	 
 	 	 	 	 	 	KernelBase.dll!SleepConditionVariableSRW()	 
 	 	 	 	 	 	realsense2.dll!Concurrency::details::stl_condition_variable_win7::wait_for(Concurrency::details::stl_critical_section_interface * lock, unsigned int timeout) Line 216	 
 	 	 	 	 	 	realsense2.dll!do_wait(_Cnd_internal_imp_t * cond, _Mtx_internal_imp_t * mtx, const xtime * target) Line 82	 
 	 	 	 	 	 	realsense2.dll!std::condition_variable::wait_for<int,std::ratio<3600,1>,<lambda_1a7ae5e86bda9bd92449f9ae1db3ee38> >(std::unique_lock<std::mutex> & _Lck, const std::chrono::duration<int,std::ratio<3600,1> > & _Rel_time, dispatcher::stop::__l2::<lambda_1a7ae5e86bda9bd92449f9ae1db3ee38> _Pred) Line 594	 
 	 	 	 	 	 	realsense2.dll!dispatcher::stop() Line 289	 
 	 	 	 	 	 	realsense2.dll!librealsense::polling_error_handler::~polling_error_handler() Line 23	 
 	 	 	 	 	 	realsense2.dll!librealsense::ds5_device::~ds5_device()	 
 	 	 	 	 	 	realsense2.dll!librealsense::rs435_device::`scalar deleting destructor'(unsigned int)	 
 	 	 	 	 	 	realsense2.dll!librealsense::pipeline::profile::~profile()	 
 	 	 	 	 	 	realsense2.dll!librealsense::pipeline::config::~config()	 
 	 	 	 	 	 	realsense2.dll!rs2_delete_config(rs2_config * config) Line 1488	 
[... cut ...]
 	 	 	 	 	 	ntdll.dll!LdrShutdownProcess()	 
 	 	 	 	 	 	ntdll.dll!RtlExitUserProcess()	 
 	 	 	 	 	 	kernel32.dll!ExitProcessImplementation()	 
 	 	 	 	 	 	msvcr100.dll!doexit(int code, int quick, int retcaller) Line 621	 
 	 	 	 	 	 	Hiris.exe!__tmainCRTStartup() Line 568	 
 	 	 	 	 	 	kernel32.dll!BaseThreadInitThunk()	 
 	 	 	 	 	 	ntdll.dll!RtlUserThreadStart()	 
```

I have to be honest, I don't know if I do something wrong or not but I implemented my dll based on example from the SDK.

dorodnic, 2019-01-28:
Hi @bansan85 
Could you please clarify what is happening when you are trying to unload the library?

bansan85, 2019-01-28:
I will try to give you a hello world with all dead lock.
Maybe this because it's a MFC library...

bansan85, 2019-01-29:
Please, clone https://github.com/bansan85/realsensedllproblem . It's a MFC DLL with only a rs2::context as global variable. Compile it.
Clone https://github.com/bansan85/realsenseloadlibrary. It's a MFC application that only do a LoadLibrary (use the right URI to the dll).

When I load the library, it's always a dead-lock.

bansan85, 2019-01-29:
Do you reproduce the problem ?

Do you need also an example when I unload the library ?

dorodnic, 2019-02-04:
Sorry, as I mentioned it can sometimes take a while. 
I compiled the app and replaced `C:\\Users\\vlegarrec\\Documents\\Visual Studio 2015\\Projects\\RealSenseProblem\\x64\\Debug\\RealSenseProblem.dll` with path to my `realsense2.dll` (and checked `pb == 0`). Closing the app did not produce a deadlock, but I'm not sure if I'm doing it right. What is in `RealSenseProblem.dll`? Is it accessing the `global_context()` from `dll_main`?

bansan85, 2019-02-04:
> What is in `RealSenseProblem.dll`?

It's the repository https://github.com/bansan85/realsensedllproblem . Please, compile it and you will have the `RealSenseProblem.dll`.
The repository https://github.com/bansan85/realsenseloadlibrary was only to show you how I load the `RealSenseProblem.dll`.

dorodnic, 2019-02-05:
@bansan85 - got it. I'm getting a deadlock in `context()` with both global and static versions. 
We will look into it, thanks

bansan85, 2019-02-05:
@dorodnic Great news. On my side, I don't have the RealSense anymore, I needed to finish the project for my customer. But please keep me in touch when it's fix so I can update his software so he can close it. Thanks.

bansan85, 2019-02-15:
Just as feedback.

I have exactly the same problem with the Euresys driver and their card "Coaxlink Octo".

I also worked with another 3D camera : the Pico Flexx. I think they had the same issue because they wrote :

>     // Windows requires that the application allocate these, not the DLL.  We expect typical
>     // Royale applications to be using a GUI toolkit such as Qt, which has its own equivalent of this
>     // PlatformResources class (automatically set up by the toolkit).

So I will keep a dynamic allocation (with new) when loading the dll and a dynamic deletion (with delete) when unloading the dll.

bansan85, 2020-06-24:
I'm cleaning up my github account.
Please find enclosed the two projects realsensedllproblem and realsenseloadlibrary. I'm no longer working on RealSense so I don't know if this bug is valid anymore.
[realsensedllproblem-master.zip](https://github.com/IntelRealSense/librealsense/files/4827651/realsensedllproblem-master.zip)
[realsenseloadlibrary-master.zip](https://github.com/IntelRealSense/librealsense/files/4827652/realsenseloadlibrary-master.zip)

RealSenseSupport, 2020-07-02:
@bansan85 Thanks for using RealSense cameras and your involvement!

closed 2020-07-02 by RealSenseSupport