realsenseai/librealsense #3140: Problems faced using D435 Camera

OthumanAnis7, 2019-01-24:
Hi,
I have a D435 camera and am trying to run some codes. But I am facing a problem of having missing libraries, files and directories and I can't seem to find the libraries online.  May I know how can I go about it?

OthumanAnis7, 2019-01-24:
Also i have problem install pyrealsense2 Library in ODROID XU4

dorodnic, 2019-01-24:
It's not very clear what "missing libraries, files and directories" means.
Did you follow the guide under [`/wrappers/python`](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source)?
What errors did you encounter?

OthumanAnis7, 2019-01-25:
Greetings,

Thank you for the prompt reply.

We are trying to run the camera in ODROID XU4, and when running all the examples which requires pyrealsense2 libraries. we can seem to install that library in ODROID XU4. Is there anny steps that we can follow in order to install that library in ODROID XU4?

Thank you.

Regards,
Anis

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

Hi @OthumanAnis7,

Do you enable the build option "BUILD_PYTHON_BINGINGS" in CMAKE? Please also review the instructions in below link. You should be able to have the library (ex: pyrealsense2.cpython-36m-xxx.so)
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

OthumanAnis7, 2019-02-21:
Greetings,
Thank you. I will try on that. Also can I know if there is any example for Intel Realsense camera D435 to run a optical flow algorithm found in GitHub?
Regards,
Anis

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi @OthumanAnis7,

I am not aware of the sample about optical flow algo + D435. @dorodnic, any info?
Or, probably, you can also check that in Intel community.
https://forums.intel.com/s/topic/0TO0P00000018NTWAY/intel-realsense-technology

closed 2019-04-19 by RealSenseCustomerSupport