realsenseai/librealsense #3141: enable/disable auto exposure

EmmanuelCos, 2019-01-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Language                            |  C++                         | 

Hello,
Does anyone know of to enable/disable auto exposure in C++?
Thank you,
Emmanuel

MartyG-RealSense, 2019-01-24:
The code for enabling and disabling the emitter can be found in Intel's RealSense SDK API how-to guide, under the 'Librealsense2' heading.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

MartyG-RealSense, 2019-01-24:
The auto-exposure can be disabled with this instruction:

      depth_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);

MartyG-RealSense, 2019-01-24:
if (depth_sensor.supports (RS2_OPTION_ENABLE_AUTO_EXPOSURE)) {

depth_sensor.set_option (RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
 
}

dorodnic, 2019-01-24:
Expanding @MartyG-RealSense answer, something like this:
```cpp
using namespace rs2;
pipeline p;
auto prof = p.start();
auto sensor = prof.get_device().first<depth_sensor>();
sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
```

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

No response in 20 days, close issue.

closed 2019-03-04 by RealSenseCustomerSupport