realsenseai/librealsense #3142: How instantiate/cast the RecordDevice C# class?

Dealerpriest, 2019-01-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC  |
| SDK Version                            | The one embedded in the unity package | 
| Language                            |  C# | 

### Issue Description
How do I get access to pause and resume when recording to bag file? I have read through [this example](https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp) from the (c++) sdk, where the device is used (casted?) as a recorder by calling `device.as<rs2::recorder>().pause()`. How could I achieve the same functionality in c#?
I've found the C# api `PlaybackDevice.FromDevice(Device dev);` which i guess achieves the same thing for playback. But how can I get the recorder? I've found the `RecordDevice` class that extends `device` and has the correct methods (pause and resume). But how in god's name do i use it?

ogoshen, 2019-01-28:
The example you linked casts a device which is **already** a recorder (created with `cfg.enable_record_to_file`)

Creating a recorder from an existing device in cpp::
```cpp
rs2::recorder recoder("ros.bag", dev);
```
and c# is pretty much the same:

```cs
RecordDevice recorder = new RecordDevice(dev, "ros.bag");
```

RealSenseCustomerSupport, 2019-02-08:
----------------------------------------------

Hi Dealerpriest,

Wonder if you have tried what ogoshen's suggested and any updates?

Thanks!

Dealerpriest, 2019-02-14:
I will try tomorrow and report my finding here then.

Dealerpriest, 2019-02-15:
Alright. I can indeed create a RecordDevice by passing a device from the DeviceList and give it a path for recording. A file will get created. It is 5kb big. But it is not filled with any data.
Here's the code:
```
using(var ctx = new Context()) {
	String recordingsPath = Application.dataPath.Replace("Assets", "recordings");

	var devices = ctx.QueryDevices();
	print("There are " + devices.Count + " connected RealSense devices.");
	if(devices.Count == 0) return;

	String recordPath = recordingsPath + "/recorder_file.bag";
	RecordDevice recordDevice = new RecordDevice(devices[0], recordPath);
	recordDevice.Resume();
}
```

I guess it's related to that the record, stream or other must be started. Do I have to create a config, pipeline and pipelineProfile anyways? And if so, how will this relate to the recordDevice?

Any advice is much appreciated!

Dealerpriest, 2019-02-15:
We have now managed to at least save some data to the bag-file. The following code is what we have at the moment:
```
void OnEnable(){
	ctx = new Context();
	ctx.OnDevicesChanged += DevicesChanged;
	String recordingsPath = Application.dataPath.Replace("Assets", "recordings");

	var devices = ctx.QueryDevices();
	print("There are " + devices.Count + " connected RealSense devices.");
	if(devices.Count == 0) return;

	String recordPath = recordingsPath + "/recorder_file.bag";
	RecordDevice recordDevice = new RecordDevice(devices[0], recordPath);
	//recordDevice.Pause();

	depthSensor = recordDevice.Sensors[0];
	//depthSensor = devices[0].Sensors[0];

	VideoStreamProfile sp = depthSensor.VideoStreamProfiles
						.Where(p => p.Stream == Stream.Depth)
						.OrderByDescending(p => p.Framerate)
						.Where(p => p.Width == 640 && p.Height == 480)
						.First();
	depthSensor.Open(sp);
	depthSensor.Start(q);
}
```

So. My understanding from reading tutorials and the docs, is that the recordDevice is a wrapper around a normal device, with the added functionality of recording (with resume and pause). After creating the recordDevice we still have to make it stream frames, which will then get saved to the bag-file. Is this correctly understood?

As of now. We will need to manually configure what sensors (and videoStreamProfiles?) will be started and thus saved to the file.
Hence, I gather it will not directly work with the playback from the unity wrapper (since there will be a discrepency of what data is expected in the bag-file)?

Is it possible to somehow control the record resume and pause, and still be sure to be able to play it back with the unity wrapper example.

ogoshen, 2019-02-17:
> So. My understanding from reading tutorials and the docs, is that the recordDevice is a wrapper around a normal device, with the added functionality of recording (with resume and pause). After [...]

Correct.

Note that you can take the shorter route, and work with the `Pipeline` by either:

1.  Using `Config.EnableRecordToFile` on the configuration object passed to the pipeline
2.  By getting the device from the `PipelineProfile.Device` returned from `Pipeline.Start` and wrapping that with a `RecordDevice`.

> Is it possible to somehow control the record resume and pause, and still be able to play it back with the unity wrapper example?

So this sounds like you're looking for point 2 mentioned above

Dealerpriest, 2019-02-18:
This seems to be exactly what I'm looking for! I will try it at the end of this week and report back. Thanks for the quick response!

Dealerpriest, 2019-02-20:
Tried the suggested method (number 2 above). Unfortunately doesn't seem to work for me. Perhaps I'm doing something wrong? If I create a pipeline with a config where ```enableRecordToFile(path)``` is used all is well and I can see the .bag file get created and grow in size. If I instead leave out the ```enableRecordToFile(path)``` and instead create a recordDevice from the device returned by ```pipe.Start(cfg)``` I won't get any data in the bag-file. Here's the code (with enableRecordToFile commented out):
```
void OnEnable()
{
	ctx = new Context();
	ctx.OnDevicesChanged += DevicesChanged;
	String recordingsPath = Application.dataPath.Replace("Assets", "recordings");

	var devices = ctx.QueryDevices();
	print("There are " + devices.Count + " connected RealSense devices.");
	if(devices.Count == 0) { return; };

	string serialNumber = devices[0].Info[CameraInfo.SerialNumber];

	pipe = new Pipeline(ctx);
	cfg = new Config();
	cfg.EnableDevice(serialNumber);

	requests = new RsVideoStreamRequest[] {
		new RsVideoStreamRequest {Stream = Stream.Depth, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Z16 , Framerate = 30 },
		new RsVideoStreamRequest {Stream = Stream.Infrared, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Y8 , Framerate = 30 },
		new RsVideoStreamRequest {Stream = Stream.Color, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Rgb8 , Framerate = 30 }
	};

	foreach(var p in requests) {
		cfg.EnableStream(p.Stream, p.StreamIndex, p.Width, p.Height, p.Format, p.Framerate);
	}

	String recordPath = recordingsPath + "/recorder_file.bag";

	//cfg.EnableRecordToFile(recordPath);

	profile = pipe.Start(cfg);

	RecordDevice recordDevice = new RecordDevice(profile.Device, recordPath);
	recordDevice.Resume();
}
```

I should mention that I can verify that a .bag file is created, but it's size is 36kb and it's not increasing in size. I have a feeling that I'm somehow not starting the stream, although I thought that was handled by the ```pipe.Start(cfg)```

Dealerpriest, 2019-02-20:
Alright!
It seems I now got it working. The issue appears to have been that I created the recordDevice inside onEnable(), which meant it went out of scope after the function was executed. When I instead declared recordDevice as a class variable, it started recording to the .bag file.

So. Lesson learned. Be very aware of scope and what object/variables need to be persistent throughout the program execution.

ogoshen, 2019-02-20:
Glad you got it figured, though you've actually did catch another bug... 
When `EnableRecordToFile` is configured, there should be a way to cast the device to a `RecordDevice` (with either `As<T>` or `RecordDevice.FromDevice`).
`PlaybackDevice` already support this, so should the recorder...

Dealerpriest, 2019-02-20:
Haha. Nice!

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi Dealerpriest,

Glad to hear that it works for you now. If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-03-14 by RealSenseCustomerSupport

snmathur4, 2019-03-18:
Can you please post the entire example for saving the depth data to .bag file, based on https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs
Thanks!

Dealerpriest, 2019-03-18:
[465494716](https://github.com/IntelRealSense/librealsense/issues/3142#issuecomment-465494716) above should be enough to get you started. Some class variable declarations are not shown but your IDE should give you warnings so you can see what declarations you must provide. For example:
```
Context ctx;
Pipeline pipe;
Config cfg;
```
etc. 
You obviously need the Intel.RealSense namespace...

The code sets up a pipeline for the first connected camera returned from `ctx.QueryDevices();`.
Also, you will need to set `recordPath` to a valid directory. The code for that above is Unity-specific.
Be aware to not let relevant variables/object get out of scope as I've already mentioned earlier above.
Alsol, I think the call to `recordDevice.Resume()` is unnecessary since it will start recording by default.

snmathur4, 2019-03-19:
Thanks, I modified the code as below, and I am able to create .bag file while streaming the depth and rgb frames.
But my stop is not working. How can I stop so the frames continue to stream? Because pipeline.Stop(); stops getting the frames completely and my program crashes. thanks

`
private Pipeline pipeline = new Pipeline();
private Colorizer colorizer = new Colorizer();
private CancellationTokenSource tokenSource = new CancellationTokenSource();

public MainWindow()
{
	InitializeComponent();

	try
	{
		var cfg = new Config();
		cfg.EnableRecordToFile("C:\temp\My_test111.bag");
		cfg.EnableStream(Stream.Depth, 640, 480);
		cfg.EnableStream(Stream.Color, Format.Rgb8);
		var pp = pipeline.Start(cfg);

		using (var p = pp.GetStream(Stream.Color) as VideoStreamProfile)
		{
			imgColor.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
			imgDepth.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
		}
		Action<VideoFrame> updateColor = UpdateImage(imgColor);
		Action<VideoFrame> updateDepth = UpdateImage(imgDepth);

		Task.Factory.StartNew(() =>
		{
			while (!tokenSource.Token.IsCancellationRequested)
			{
				using (var frames = pipeline.WaitForFrames())
				{
					var colorFrame = frames.ColorFrame.DisposeWith(frames);
					var depthFrame = frames.DepthFrame.DisposeWith(frames);
					var colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames);

					// Render the frames.
					Dispatcher.Invoke(DispatcherPriority.Render, updateDepth, colorizedDepth);
					Dispatcher.Invoke(DispatcherPriority.Render, updateColor, colorFrame);
				}
			}
		}, tokenSource.Token);
	}
	catch (Exception ex)
	{
		MessageBox.Show(ex.Message);
		Application.Current.Shutdown();
	}
}

private void control_Closing(object sender, System.ComponentModel.CancelEventArgs e)
{
	tokenSource.Cancel();
}

static Action<VideoFrame> UpdateImage(Image img)
{
	var wbmp = img.Source as WriteableBitmap;
	return new Action<VideoFrame>(frame =>
	{
		using (frame)
		{
			var rect = new Int32Rect(0, 0, frame.Width, frame.Height);
			wbmp.WritePixels(rect, frame.Data, frame.Stride * frame.Height, frame.Stride);
		}
	});
}

private void Button_Click_Stop(object sender, RoutedEventArgs e)
{
	pipeline.Stop();
}
`