realsenseai/librealsense #3143: The issue for running examples

lfywork, 2019-01-24:
Capturing DEPTH at 640 x 480, fov = 67.5 x 53.3, distortion = INVERSE_BROWN_CONRADY
Capturing COLOR at 640 x 480, fov = 54.7 x 42.4, distortion = NONE
Capturing INFRARED at 640 x 480, fov = 67.5 x 53.3, distortion = INVERSE_BROWN_CONRADY
rs.error: TemperatureControlLoop: opcodes do not match
RealSense error calling rs_start_source(device:0x22c6c40, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
Press <RETURN> to close this window...

do you know how to resolve it ?THANKS YOU.

lfywork, 2019-01-24:
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
unit-tests-live-f200 is a Catch v1.2.1 host application.
Run with -? for options

-------------------------------------------------------------------------------
F200 streams depth (Z16)
  F200 streams depth 640x240 (HVGA), [15,30,60] fps
-------------------------------------------------------------------------------
../unit-tests/unit-tests-live-f200.cpp:146
...............................................................................

../unit-tests/unit-tests-live-f200.cpp:164: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  Testing 15 fps
  SIGSEGV - Segmentation violation signal

===============================================================================
test cases:    4 |    3 passed | 1 failed
assertions: 2367 | 2366 passed | 1 failed

Press <RETURN> to close this window...

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

Hi @lfywork,

Which f200 firmware do you use? which app in librealsense1 v1.12.1 do you run to have this error?

lfywork, 2019-02-28:
Hi @RealSenseCustomerSupport,
Thank you for your attention. I solve it by myself. Before update ubuntu kernel, I run the examples successfully. So I doubt that kernel version is too high. When I use the low version kernel, I run app successfully.

closed 2019-02-28 by lfywork