realsenseai/librealsense #3144: Issue with IR sensor/image

ajprax, 2019-01-24:
**Camera Model**: D415
**Firmware Version**: 5.10.13
**Operating System & Version**: MacOS 10.14.2
**Platform**: PC
**SDK Version**: 2.2.17
**Language**: python
**Segment**: Robot

### Issue Description
I've had an issue where depth quality degrades periodically and it has become more frequent over time. My depth images have increasingly common 0 depth readings scattered randomly throughout the image and depth precision is reduced. I think I've narrowed the issue to the IR sensor. I looked at the IR stream using realsense viewer when things were working well, and looking at it now, it looks very different. It's a little hard to describe, but before the image was more stable (now it is very noisy) and the colors are different (before it was a kind of dark grey or purple, now brighter green and red). Unfortunately I don't have a "before" image to include, but here's what it looks like now:
[image]
I haven't (knowingly) changed any settings when the issue manifests.

MartyG-RealSense, 2019-01-25:
Does the image improve at all if you turn off post-processing, please?

ajprax, 2019-01-25:
I had post-processing off in the viewer for all of these screenshots. When I'm capturing frames from the SDK in python the only post-processing I've been doing is `rs.align`.

MartyG-RealSense, 2019-01-25:
Apologies for the delay in responding.   I have been considering every scenario I can think of that might fit your symptoms.

Your camera temperature seems to be within acceptable operating ranges.  So unless the temperature is rising as time goes on during streaming, it probably isn't a temperature-related instability.

The Intel support guys on this forum may be able to offer a good explanation.  My personal thoughts are edging towards a couple of possibilities to account for why the problem develops over time instead of at the start of the stream:

1.  An instability in your USB port.  Do you have another USB port on your computer that you can try the camera in?

Or 2.  If you have written your own program, there is something in the code's structure that is causing resources to not be released properly after each frame when it loops round to capturing the next frame.  So over time, performance may get increasingly worse as the computer's memory is consumed.

RealSenseCustomerSupport, 2019-02-01:
----------------------------------------------

Hi

- From your Infrared1 camera image, I don't think such pattern noise is related to temperature. It looks some certain registers setting not correct, or sensor itself became defective now.
- Do you see same issue on Infrared 2? (although in Y8 mode, such noise pattern should be visible).
- If No, probably the Infrared1 sensor is defective now. If Yes, can you try other format like 640x360 ....? see same issue?
- If Yes, can you try firmware update again? (can be one version earlier or newer).
- And check camera again with RealSense Viewer. Run Infrared1 camera only under same format: RGB8 and 1280x720 30FPS. If the issue is still there, I guess this camera sensor is defective now.

Thanks

RealSenseCustomerSupport, 2019-02-09:
----------------------------------------------

Hi

Is there any update for this issue?

Thanks

ajprax, 2019-02-09:
Unfortunately I haven't had a chance to try these things, I've moved on to other parts of the project with a workaround to identify and skip frames with bad depth data. I'll likely come back to this eventually, but you can close it for now if you want to keep things tidy.

RealSenseCustomerSupport, 2019-02-09:
----------------------------------------------

Thank you for the status update. I'd appreciate if you can share the root cause with us if you get it fixed.
OK, I'm going to close it for now and you're always welcome to re-open it in the future.

Thanks and have a good weekend

closed 2019-02-17 by RealSenseCustomerSupport

tonyzzzzz, 2019-03-07:
i saw the same pattern noise on my D415, which I just received today. It looks like the noise is caused by the embedded IR LED. i.e., if I cover the internal LED and use an external IR to illuminate the object, the patter noise went away. Can anyone help on this? btw - I did upgraded my D415 to the latest firmware.

[image]

MartyG-RealSense, 2019-03-07:
The pattern on your image is consistent with one produced by the camera's IR Emitter component.  I took a snapshot of myself to illustrate this.

[image]

The IR Emitter projects a pattern on semi-random dots onto objects in a scene.  This helps to add analyzable texture to low-texture objects such as doors and walls.  The emitter can have its Laser Power turned down to reduce the pattern's visibility or be turned off altogether.

The emitter can be altered with scripting.  If you are using the RealSense Viewer though, you can change the Laser Power setting or disbale / enable the emitter in the options side-panel

To turn the emitter off in the Viewer, left-click on the blue box beside the 'Emitter Enabled' option so the box turns to black, indicating that it is turned off.  The image then resembles the one you captured with external IR.

[image]

The Laser Power setting can be found in the Viewer by expanding open the 'Controls' sub-option in the 'Stereo Module' section of the side-panel.

[image]

tonyzzzzz, 2019-03-08:
@MartyG-RealSense thanks! appreciate.