realsenseai/librealsense #3145: E: Unable to locate package librealsense2-dkms

lz130dd, 2019-01-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 16) }  | 
| Kernel Version (Linux Only)    |  (4.4.38)                                         | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When I follow the instructions in the file "distribution_linux.md", after I run the commend

$ sudo apt-get install librealsense2-dkms

I got error like this,

$ sudo apt-get install librealsense2-dkms 
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package librealsense2-dkms

I have tried to use " sudo apt-get update", but it doesn't work. Can anybody help me with this?

dorodnic, 2019-01-24:
Hi @lz130dd 
We do not provide pre-built modules for non-x86 architectures. 
There is, however, plenty of information on this site and [jetsonhacks.com](https://www.jetsonhacks.com/) to help get the camera working on Jetson.

closed 2019-02-17 by RealSenseCustomerSupport

kidier, 2019-10-15:
Help me.
i have got same case:
$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package librealsense2-dkms

I tried solve that problem from existing solves on forum.. unsuccesfully.
My hardware:  Up-Board Ubuntu 18.04

nimisha3005, 2023-06-04:
@kidier were you able to solve the issue?