realsenseai/librealsense #3147: Pre-recording Bag files

kabawo, 2019-01-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0              | 
| Platform                                 | PC |
| SDK Version                            |     2.17.1             | 
| Language                            |  OpenCV/Python                          | 
| Segment			|  others - Visual SLAM                          | 

Hello,
I am recording color and depth video using realsense-viewer, which results in producing a .bag file. I am not using ROS, but pure Python/OpenCV, so I am struggling to consume the pre-recorded information. I am using a tool from the following repository  [https://github.com/UnaNancyOwen/rs_bag2image](https://github.com/UnaNancyOwen/rs_bag2image) to 'unpack' the .bag file.
This gives me color and depth images - color as jpg and depth as png.
I can re-create a video using ffmpeg (which I did for RGB). However, my concern is regarding how to correctly process the depth data.
I will be using CNN on color image to find specific pixel points of interest (these could be whole lines or only few points per frame) and I need to be able to read the depth information of those pixels.
Is this a correct approach or should the depth channel be in RAW format? If so, how would I record it in such a format? Reading through other issues (e.g. #2231) it seems the PNGs I have extracted will not be useful and I need RAW files, which I would then align with RGB. Could I achieve this alignment in Python or do I have to pre-process both streams using rs_align?
Alternatively, maybe I can read both streams from the .bag file directly in Python? If that is the case, could someone point me towards a code sample?
Thanks for reading such a lengthy description and please, let me know if I need to provide any further details or code samples.

dorodnic, 2019-01-25:
Hi @kabawo 
Well, consuming the bag file directly from python would be the easiest:
```python
p = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file("bag.bag")
prof = p.start(cfg)
```
You can then set `real_time_mode` to false:
```py
prof.get_device().as_playback().set_real_time(false)
```
And fetch all the frames, just like you would from a device:
```py
fs = p.wait_for_frames()
d = fs.get_depth_frame()
```
You could also do alignment directly in python:
```
a = rs.align(rs.stream.color) // Make sure to do this only once and not for each frame
...
fs = a.process(fs)
```
Eventually if you do want to store depth image, any lossy encoding is a bad idea.
Not simply because you'd loose quality, but because depth doesn't work like RGB - interpolating between an object and a wall behind it does not make a lot of sense.
Any lossless compression will do. You can use lz4, and even use PNG. The problem most people have with storing depth as PNG is when you do it after colorisation (=histogram equalisation). That's a destructive operation, there is no way back. But you could store depth as 16-bpp greyscale PNG (using opencv for instance from python)

ogoshen, 2019-01-28:
`enable_device_from_file` has a second optional parameter named `repeat_playback` that defaults to true.

kabawo, 2019-01-28:
> `enable_device_from_file` has a second optional parameter named `repeat_playback` that defaults to true.

Thanks. This works. The only problem is, I have a 'while' loop to embed the streams in PyGame and 
`frames = pipe.wait_for_frames()`
still waits for frames after the end of playback, which ends in 'RuntimeError: Frame didn't arrive within 5000'.

Are you aware of any way of recognizing the end of playback? Thanks

ogoshen, 2019-01-29:
You can use `try_wait_for_frames`  instead which returns tuple `(success, frames)`
Also something like to detect playback status:
```python
playback = rs.playback(profile.get_device())
while playback.current_status == rs.playback_status.playing:
    # grab and process frames
    pass
```

kabawo, 2019-01-31:
> You can use `try_wait_for_frames` instead which returns tuple `(success, frames)`
> Also something like to detect playback status:
> 
> ```python
> playback = rs.playback(profile.get_device())
> while playback.current_status == rs.playback_status.playing:
>     # grab and process frames
>     pass
> ```

That would be exactly what I need, but I get these:
`AttributeError: 'pyrealsense2.pirealsense2.pipeline' object has no attribute 'try_wait_for_frames'`

I also struggled with the second solution, before I added the brackets ():
`while playback.current_status() == rs.playback_status.playing:`
in case someone else needs it.

Thanks for these great pointers. I really appreciate the help.

closed 2019-01-31 by kabawo