realsenseai/librealsense #3148: Intel RealSense Viewer on Windows 10 doesn't respond to mouse...

drhalftone, 2019-01-25:
I run the RealSense viewer on my Windows 10 machine.  The software recognizes my D435, showing the toggle buttons for Stereo as well as RGB.  However, no matter what I try to click on with my mouse, nothing happens.  Its as if I took a screen shot of the window, and then opened the screen shot in Photoshop and just clicked on the raster image.

What gives?

drhalftone, 2019-01-26:
Is there a solution for this?

MartyG-RealSense, 2019-01-26:
It is difficult to diagnose a cause if the entire program is freezing on start-up, and it is not just a case of it freezing when the stream is activated.

Could you try opening the Viewer without the camera plugged in, please?  You won't have much that can be interacted, but there are '2D', '3D' and the gear-wheel setup icon buttons in the top corner.

[image]

Could you try clicking on them please when the camera is not attached and see if they respond?  If they do, this would suggest that the freezing problem begins when the camera is detected and the options side-panel is generated.

dorodnic, 2019-01-27:
Hi @drhalftone 
I have never encountered this. Did you try one of the older stable releases (such as [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)
If it reproduces there as well, most likely something is wrong with the OpenGL subsystem on your machine. If it does not, this can be a new bug. Either way, we need more information about your system - what is the Windows version and build, is this a laptop / PC / surface or some other type of Windows device?

closed 2019-02-17 by RealSenseCustomerSupport