realsenseai/librealsense #3150: Exposure not supported for color stream on D435i?

mkaspr, 2019-01-26:
Hello,

I've recently switched from a D435 to a D435i and have noticed that manually setting exposure on the color stream is no longer supported on the D435i. That is, the following function call returns false on the D435i:

`my_color_stream.supports(RS2_OPTION_EXPOSURE)`

and

`my_color_stream.supports(RS2_OPTION_ENABLE_AUTOEXPOSURE)`

The same code works fine when using the D435 and even with IR streams on the D435i. Is this a potential bug or a limitation of the device?

Thank you,
Mike

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.17.1                         | 
| Language                            |  C++                          | 
| Segment			| Robotics                         |

RealSenseCustomerSupport, 2019-02-23:
----------------------------------------------

Hello @mkaspr,

Thank you for your inquiry.  A couple questions:

This issue may be OS specific, would you be able to try your code on a Windows 10 system?

Also, have you updated your firmware to latest?
https://downloadcenter.intel.com/download/28481/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128255

Linux Device Firmware Update Tool:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

Also try using the latest LibRealSense Release:
https://github.com/IntelRealSense/librealsense

-  Please try this on Windows 10 with updated FW + LibRealSense Binaries
-  Also try updated FW + LibRealsense Binaries on Linux Ubuntu 16.04 as well.

Update this ticket with your results and we can proceed from there.

Thank you!

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hello

Is the issue still seen on your side?
Under Win10, with RealSense Viewer(2.19.0) and FW5.11.1.100, I don't see the issue on manual exposure control on RGB camera of D435i.

Thanks

closed 2019-04-10 by RealSenseCustomerSupport