realsenseai/librealsense #3151: Toggling the IR emitter while video is playing...

drhalftone, 2019-01-26:
Are there any examples of how to toggle the IR emitter while the camera is running?  I don't want to toggle on every frame.  I just want to do what RealSense viewer does which is to toggle the IR emitter at any time.  The only way I have been able to do it, is completely shutdown the video, toggle the emitter, and then reconnect and launch the camera streams again.  

Is there any example code?

MartyG-RealSense, 2019-01-26:
The link below has C++ code for controlling the emitter on-off state, under the 'Librealsense2' heading.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

There is also a recent example from Dorodnic the RealSense SDK Manager for controlling auto-exposure on / off that you could probably adapt to use the emitter control instruction.

https://github.com/IntelRealSense/librealsense/issues/3141#issuecomment-457361895

drhalftone, 2019-01-26:
What about C code instead of C++?

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser <https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser>

#3141 (comment) <https://github.com/IntelRealSense/librealsense/issues/3141#issuecomment-457361895>

MartyG-RealSense, 2019-01-26:
The only reference I could find was a C user who said that they set the emitter manually using set_option.

https://github.com/IntelRealSense/librealsense/issues/1217

I do not know the C language well enough to explain how it works though.  One of the Intel guys such as Dorodnic may be able to help with that.

drhalftone, 2019-01-26:
Yes, that post describes how I set the emitter state before I start streaming video.  So the problem now is how do I change the state after I've started the pipeline, and as you suggest, I'll put this on the back burner until the Intel team can get back to me.

dorodnic, 2019-01-27:
It would be roughly the same, just like with python, Matlab and other wrappers. 
We do provide couple of simple [C examples](https://github.com/IntelRealSense/librealsense/tree/master/examples/C).
C bindings exist mostly for integration with other languages, so I'd strongly advice against this approach.

drhalftone, 2019-01-29:
Solution:
`        // CREATE A VARIABLE TO KEEP TRACK OF ERRORS
        rs2_error *err = nullptr;
        rs2_device *device = rs2_pipeline_profile_get_device(profile, &err);
        if (err) {
            errorString = QString("Unable to retreive device from profile: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        }

        // GET SENSOR LIST
        rs2_sensor_list *sensorList = rs2_query_sensors(device, &err);
        if (err) {
            errorString = QString("Unable to retreive valid context: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        }

        // MAKE SURE WE HAVE AT LEAST ONE VALID SENSOR
        int numSensors = rs2_get_sensors_count(sensorList, &err);
        if (err) {
            errorString = QString("Unable to retreive valid context: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        } else if (numSensors == 0) {
            errorString = QString("Unable to retreive valid context: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        }

        // GET A HANDLE TO THE FIRST SENSOR IN THE LIST
        rs2_sensor *sensor = rs2_create_sensor(sensorList, 0, &err);
        if (err) {
            errorString = QString("Unable to create sensor: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        } else if (sensor == nullptr) {
            errorString = QString("Unable to retreive valid sensor.");
            emit emitError(errorString);
            return;
        }

        // ENABLE/DISABLE THE INFRARED EMITTER
        rs2_set_option((rs2_options *)sensor, RS2_OPTION_EMITTER_ENABLED, (float)state, &err);
        if (err) {
            errorString = QString("Unable to disable emitter: %1").arg(QString(rs2_get_error_message(err)));
            emit emitError(errorString);
            return;
        }

        // DELETE THE SENSOR NOW THAT WE ARE DONE WITH IT
        rs2_delete_sensor(sensor);
        rs2_delete_sensor_list(sensorList);
        rs2_delete_device(device);
`