realsenseai/librealsense #3153: Is there any existing software for measuring the distance between two points?

czjczjczj1, 2019-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.16.3                           | 
| Language                            |  python                      | 
| Segment			|  others                      | 

### Issue Description
Hello, I saw a youtube video where someone used a software doing something like measuring the distance two points. This is the youtube link: https://www.youtube.com/watch?v=9gO30MG89Qk. At 7:14, it seemed that he was measuring the distance between points. So I am wondering is there any software like this?

Core055, 2019-01-26:
rs-measure within the librealsense
[image]

dorodnic, 2019-01-27:
Of course, `rs-measure` is just an example that is intended to give you some ideas.
For the time being, we try to steer clear from developing production level software tools, giving more space to the community and [RealSense ISVs](https://realsense.intel.com/software-for-intel-realsense/).
However, we might publish a more well-rounded measurement tool sometimes in the future to better illustrate hardware capabilities.

closed 2019-02-20 by RealSenseCustomerSupport