realsenseai/librealsense #3154: Intel Realsense Viewer app

Core055, 2019-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   { MacOS } | 
| SDK Version                            |  { legacy / 2.<16>.<1> }                          | 
| Language                            |  {C/C++}                          | 

I want to make a new app by reducing the Intel Realsense Viewer app,
[image]
how can I move this to the top

Core055, 2019-01-26:
#librealsense 2.16.1

dorodnic, 2019-01-27:
Hi @Core055 
The [Depth Quality Tool](https://github.com/IntelRealSense/librealsense/blob/master/tools/depth-quality/rs-depth-quality.cpp) is doing something similar. 
However, SDK tools are provided as reference code, not ready to use UI building blocks, so in order to accomplish this, you will need to dive deeper into the code.

MartyG-RealSense, 2019-01-27:
Hi Dorodnic, this question was also posted on Intel Support and has been marked as answered.  Thanks!

https://forums.intel.com/s/question/0D50P00004D1axFSAR/i-want-to-make-a-new-app-by-reducing-the-intel-realsense-viewer-appusing-c-librealsense-2161how-can-i-move-this-to-the-topthankyou?language=en_US

closed 2019-02-17 by RealSenseCustomerSupport