realsenseai/librealsense #3155: Aligning frames from separate rosbag files after HW sync

calebweinreb, 2019-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-43-generic  | 
| SDK Version                            |  2.16.5    | 
| Language                            |  python                          | 
| Segment			|  Research       | 

### Issue Description
I am recording from 5 realsense cameras simultaneously (D415), getting depth and color at 640 X 480. I have them hardware synced and have set one as master and the rest as slave. My goal is be able to align frames from the 5 separate rosbag files created by each camera. Currently, with or without HW sync, I can get them aligned to within roughly 10-30ms by using the software timestamp and cross-referencing it to a common clock on the computer. However, I would like the cameras to be perfectly synchronized. Using a panel of LEDs that that generates a different pattern every 4ms, I can see that my current system fails to precisely align the frames (even with HW sync). What is the correct way to go about this? I am using pyrealsense2.

MartyG-RealSense, 2019-01-26:
The multiple camera white paper states that if hardware sync is working correctly then you will see the timestamps drift over time.

The Chief Technical Officer (CTO) of the RealSense Group at Intel gives some excellent advice in the link below as part of a much larger discussion that is well worth reading from the beginning.

https://intel-openport-v8.hosted.jivesoftware.com/message/593138#593138

calebweinreb, 2019-01-29:
Thanks! The problem for me is actually a little more basic than that: I'm not sure how to align frames from separate rosbags at all in the first place. I had a pipeline that worked fine (to within 20ms) when I was doing realtime playback from the rosbags, but that approach is no longer working with non-realtime playback. 

Currently my pipeline involves recording from several cameras simultaneously by looping over pipelines and calling ```wait_for_frames()```. During the first several loop iterations, I record timestamps in a separate file to be used later for offsetting each cameras clock. 

Then, during playback, I call ```playback.set_real_time(False)``` and once again loop over pipelines calling ```wait_for_frames()```. At this point, I would like to get the original timestamps from each frame for alignment, but I'm confused how to do this. ```rs.frame.get_timestamp(frames)``` does not give a consistent answer for the same frames each time I run it. I was also considering using ```rs.frame.get_frame_number(frames)```, but I am concerned that this will fail if there are dropped frames from the original recording and thus gaps in the rosbag file.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi calebweinreb,
Just want to add that there are different types of timestamps that are available in the frame metadata.

The following two are based on hardware device clock:

    RS2_FRAME_METADATA_FRAME_TIMESTAMP                      , /**< Timestamp set by device clock when data readout and transmit commence. usec*/
    RS2_FRAME_METADATA_SENSOR_TIMESTAMP                    , /**< Timestamp of the middle of sensor's exposure calculated by device. usec*/

The following two are based on the system host clock:

    RS2_FRAME_METADATA_TIME_OF_ARRIVAL                      , /**< Time of arrival in system clock */
    RS2_FRAME_METADATA_BACKEND_TIMESTAMP                    , /**< Timestamp get from uvc driver. usec*/

closed 2019-03-06 by RealSenseCustomerSupport