realsenseai/librealsense #3156: OSX + brew :: Failure to build from github instructions

jacksonkr, 2019-01-27:
```
Operating System & Version MacOS 10.14
Homebrew 1.9.3
SDK Version 2
cmake 3.12.3
```

### Issue Description

When trying to build via the [github osx instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md) I run into the following issue. What's going on?

```
$ cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode
CMake Error:
  Xcode 1.5 not supported.

CMake Error: Could not create named generator Xcode

Generators
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Xcode                        = Generate Xcode project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
```

drhalftone, 2019-01-29:
Drop the XCode option.  Just do the make command followed by make -j8 and then sudo make install.  Works like a charm.

jacksonkr, 2019-01-29:
@drhalftone That's ultimately what I ended up doing but the build left me a little dumbfounded as I expected some sort of `bin` folder or at least a set up executables inside the new folders. Am I missing something obvious?

drhalftone, 2019-01-29:
Only thing I can think of is maybe you updated XCode without installing the command line tools.  Did you run "xcode-select —install”

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@jacksonkr Did you get it through? Looking forward to your update. Thanks!

jacksonkr, 2019-02-11:
I switch to linux to get the sdk to work as that was ultimately the easier option. One day I hope to get back to this but now I guess it will have to be closed :/

closed 2019-02-11 by jacksonkr

HosikChae, 2020-10-18:
I got the same error, which comes from that CMake cannot the Xcode installed, and it was because I moved to the Xcode to another place from '/Application/Xcode'. I moved back the Xcode to its initial path, and then it worked.