realsenseai/librealsense #3158: API version mismatch: librealsense.so was compiled with API version 2.18.0 but the application was compiled with 2.17.1!

kracon7, 2019-01-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.10.06.00| 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15         | 
| Platform                                 | PC|
| SDK Version                            |   2.18.0                        | 
| Language                            |                        | 
| Segment			|  Robot      | 

### Issue Description
Try to do ROS launch realsense2_camera and got an error.
 It worked previously. I don't know what caused this problem.
Thanks in advance!!!

Error message as follows:

[ERROR] [1548646771.535195820]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.18.0 but the application was compiled with 2.17.1! Make sure correct version of the library is installed (make install)
[ERROR] [1548646771.535220097]: The error before refreshing the cache was: API version mismatch: librealsense.so was compiled with API version 2.18.0 but the application was compiled with 2.17.1! Make sure correct version of the library is installed (make install)
[FATAL] [1548646771.535390854]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 27815, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/jc/.ros/log/4983a916-22ac-11e9-bc97-9cb6d0bcf2df/camera-realsense2_camera-2.log].
log file: /home/jc/.ros/log/4983a916-22ac-11e9-bc97-9cb6d0bcf2df/camera-realsense2_camera-2*.log

dorodnic, 2019-01-28:
Based on the error description, you have librealsense 2.18.0 installed, but older version of ROS-realsense.
Can you try [intel-ros/realsense/releases/2.1.4](https://github.com/intel-ros/realsense/releases/tag/2.1.4)?

kracon7, 2019-01-28:
It's working! Thanks a lot!

Jiacheng

closed 2019-02-08 by RealSenseCustomerSupport