realsenseai/librealsense #3160: How to Adapt Sample Python Code for my Needs

ShadowHunter0510, 2019-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v2.17.0 | 
| Operating System & Version | Ubuntu 16 | 
| Kernel Version (Linux Only)    | 4.19.8-v7+ | 
| Platform                                 | Raspberry Pi |

### Issue Description

I am currently trying to utilize the basic functionality of the camera, tracking the distance from the object to the user, with the code outlined in [distance_to_object.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb). However, instead of attaching a prerecorded video into the program, I want to take the live video from the RealSense Depth Camera and track the distances in live time. @dorodnic suggests that “remove the line cfg.enable_device_from_file("../object_detection.bag") [which] makes it run from previously collected recording.” Once it’s removed, he claims the demo will try to connect to a live camera. Can someone elaborate why this would work and how I would fine tune it for my needs? I appreciate your help!

MartyG-RealSense, 2019-01-28:
I believe that if configuration (cfg) instructions are not provided at the start of a script then the camera uses its default configuration.

ShadowHunter0510, 2019-01-28:
Hi @MartyG-RealSense 

I”ve tried running about half of the code, and it executes perfectly, but MatPlot isn’t working as intended since no image is being outputted. Adding “plt.show()” after each of the imshow’s did not work either. I know that this is more of a MatPlot issue rather than an Intel RealSense Camera issue, but I believe it might be because the code is being forced to use the default configuration, as you mentioned, rather than an implemented image. Would I need to change anything in the code for my specific needs?

MartyG-RealSense, 2019-01-28:
When you say Matplot, do you mean the Python plotting library Matplotlib, please?

I located an Intel tutorial for Python where Matplotlib is referenced.

https://colab.research.google.com/drive/10YTLAf2i0R80-XX_6-1gTPtOMD0cRIx5

In that guide, Matplotlib is imported with the following instruction:

**import matplotlib.pyplot as plt**

ShadowHunter0510, 2019-01-28:
Hi @MartyG-RealSense 

Yes, that is what I mean. I am only asking about this because in distance_to_object.ipynb, the code outputs an image after .imshow(), but mine does not. I am only afraid that the code might not be reading the input from the camera properly.

MartyG-RealSense, 2019-01-29:
RealSense programming is not one of my specialist areas, so one of the Intel guys such as Dorodnic can give a better answer.

If you want something to try in the meantine though, you could try replacing the line that calls the bag file:

cfg.enable_device_from_file("../object_detection.bag")

with this pair of cfg instructions for color and deph streams.

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

***********

To change the resolution and FPS in these instructions, change the first two numbers in the bracket to the resolution you want to use  - for example, for 1280x720 is 1280, 720 - and the final number from 30 to the FPS value you prefer to use.  Of course, the resolution and FPS combinations have to be ones that are supported by the camera.

I have to go to bed now unfortunately.  I wish you best of luck!

ShadowHunter0510, 2019-01-29:
If I may add,

I have tried copying and pasting the code as well to see what happens, and I receive an error code in regards to OpenCV. I believe that this line in particular converts the perceived distances to the correct units and that there’s a workaround? Again, if this is only for detecting what an object is, I do not need that. I only need the distances. [image]

MartyG-RealSense, 2019-01-29:
You could make it detect distance without displaying the object name by editing the final line to take out the instruction to display the variable containing the object name.  For example, change it to:

print("The object is {1:.3} meters away.".format(className, dist))

It is an inelegant solution because the program is still doing object recognition calculations in the background, but in programming terms it is easier than pulling out bricks and risking having the whole house crash down.

RealSenseCustomerSupport, 2019-02-14:
----------------------------------------------

Hi @ShadowHunter0510,

Do you resolve the issue? You can base on @MartyG-RealSense suggestion to have the distance detection.
By the way, you need to download "**MobileNet SSD Model" in order to avoid the open error if you like to have the object detection.**

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

I will close this. If any further question, you can re-open or create another ticket.

closed 2019-03-10 by dorodnic