realsenseai/librealsense #3164: Calling keep() produces memory leak in Python

krejov100, 2019-01-28:
D400, OS independent, PC/UpBoard, Python

This is an issue that follows up to the following merged pull request #1015. There needs to be a mechanism added to handle the release of the data once its use is finished with. I am not able to find a python call to release said kept frame, resulting in a severe memory leak which garbage collection is not able to handle. Having looked through rs_frame.hpp, I believe there needs to be a corresponding call to rs2_release_frame which is currently done in the C++ destructor.

lramati, 2019-01-30:
I can't recreate this behavior.
```python
import pyrealsense2 as rs

pipe = rs.pipeline()
pipe.start()

df = rs.decimation_filter()

while (True):
    f = pipe.wait_for_frames()
    f.keep()
    d = f.get_depth_frame()
    if d:
        f2 = df.process(d)
        f2.keep()
```
stabilizes memory-wise at < 50MB on both Python 2 and Python 3.6 in debug and release modes

dorodnic, 2019-01-30:
@krejov100 - I'd like to double check the issue is real before merging #3174 
Perhaps frames are being queued somewhere else in your project?

krejov100, 2019-01-30:
Yep so I found the issue was in fact pickling the numpy array. Numpy recently had a regression in 1.16.0
https://github.com/numpy/numpy/issues/12793
reverting to 1.15.4 means I no longer see memory leaks.

closed 2019-01-30 by krejov100