realsenseai/librealsense #3165: No device detected. Is is plugged in? realsense-viewer 2.18.0

sophie-shin, 2019-01-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.18.0 | 
| Operating System & Version |   Jetson TX2 ubuntu 16.04  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Jetson TX2  |
| SDK Version                            |                            | 

### Issue Description
I am trying to use realsense with Jetson TX2 (jetpack 3.2.1).

I took 6 days to connect a realsense D435i.. but still failed.. please help me.

I followed below to install the realsense lib.

1. Install Jetpack 3.2.1 with host PC.

2. Install realsense as below.

git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git branch v2.9.1
git checkout -b v2.9.1
sudo apt-get update
sudo apt-get install git cmake
sudo apt-get install libssl-dev
sudo apt-get install libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install libglfw3-dev
sudo apt-get install libudev-dev
sudo apt-get install cmake-curses-gui
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
reboot
Build and Install librealsense
mkdir build && mkdir install
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")
cd build
cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DBUILD_UNIT_TESTS=false
make -j4
sudo make install

I tried another installation guide such as Jetsonhacks, but it also shown no devices detected.

nvidia@tegra-ubuntu:~$ rs-enumerate-devices
No device detected. Is it plugged in?
nvidia@tegra-ubuntu:~$ usb-devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.04
S:  Manufacturer=Linux 4.4.38-tegra xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=3530000.xhci
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=05e3 ProdID=0608 Rev=32.98
S:  Product=USB2.0 Hub
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=413c ProdID=2107 Rev=01.15
S:  Manufacturer=Dell
S:  Product=Dell USB Entry Keyboard
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=1.5 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=03f0 ProdID=0b4a Rev=72.00
S:  Manufacturer=Logitech
S:  Product=USB Optical Mouse
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 3
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev=04.04
S:  Manufacturer=Linux 4.4.38-tegra xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=3530000.xhci
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  9 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0b3a Rev=50.ad
S:  Manufacturer=Intel(R) RealSense(TM) Depth Camera 435i
S:  Product=Intel(R) RealSense(TM) Depth Camera 435i
S:  SerialNumber=850123051723
C:  #Ifs= 6 Cfg#= 1 Atr=c0 MxPwr=440mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 5 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

sophie-shin, 2019-01-29:
nvidia@tegra-ubuntu:~$ modinfo uvcvideo | grep ver
filename:       /lib/modules/4.4.38-tegra/kernel/drivers/media/usb/uvc/uvcvideo.ko
version:        1.1.1
description:    USB Video Class driver
srcversion:     0CD1BCB0F0E7ED2CADBCA5D
vermagic:       4.4.38-tegra SMP preempt mod_unload aarch64

sophie-shin, 2019-01-29:
nvidia@tegra-ubuntu:~$ uname -r
4.4.38-tegra

sophie-shin, 2019-01-29:
nvidia@tegra-ubuntu:/usr/local/bin$ ./realsense-viewer
 29/01 04:25:59,599 WARNING [547770490880] (ds5-factory.cpp:578) DS5 group_devices is empty.

MartyG-RealSense, 2019-01-29:
Have you seen the official documentation page for installing Librealsense on Jetson Tx2 and performed the steps recommended in that guide, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

dorodnic, 2019-01-29:
This part:
```
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git branch v2.9.1
git checkout -b v2.9.1
```
Doesn't look right. v2.9.1 was release long before D435i was a product. 
JetsonHacks just published a tutorial on the D435i: [intel-realsense-d435i-on-nvidia-jetson-agx-xavier/](https://www.jetsonhacks.com/2019/01/21/intel-realsense-d435i-on-nvidia-jetson-agx-xavier/)
Perhaps it could help

sophie-shin, 2019-01-29:
@MartyG-RealSense

Thank you for the comment. OF COURSE! I checked the official site first. and my windows laptop already set up based on the information provided by official site. However, the install guide for Jetson TX2 is not existed. So, I follow another install guide such as Jetsonhacks and etc. I reset Jetson TX2 board several time with Jetpack 3.2.1 and Jetpack 3.3 before following the install guild for removing any interruption. 

And the official documentation page what you mentioned.. said, edit ./scripts/patch-realsense-ubuntu-xenial.sh. BUT, I can't find patch-realsense-ubuntu-xenial.sh in /librealsense/scripts dir. 

Please help.

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Freeman Lo, Jan 31, 3:17 PM

Hi @sophie-shin,

For TX2, refer to the build process in below link. In order for librealsense to work properly, the kernel Image must be rebuilt and patches applied to the UVC module and some other support modules. In addition, for support of the extra features of the D435i camera such as the IMU, extra modules must be built. These modules are in the Industrial I/O ( **IIO** ) device tree. The Jetson already has IIO support enabled in the kernel image to support the INA3321x power monitors. To support these other HID IIO devices, IIO_BUFFER must be enabled; it must be built into the kernel Image as well.
https://github.com/jetsonhacks/buildLibrealsense2TX

Make sure to run the script "buildPatchedKernel.sh" correctly. Then, above kernel patches especially for HID IIO devices should be applied. If this is ready, you should be able to run D435i video and IMU streams.

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Freeman Lo, Jan 31, 3:18 PM

Hi @sophie-shin,

Make sure to use the latest librealsense like 2.18.0.

sophie-shin, 2019-02-08:
Thank you all,

buildPatchedKernel.sh fixed it correctly.

closed 2019-02-08 by sophie-shin

228282416, 2019-05-20:
This solution is out of date.So,how can I install D435i driver on Tx2(Jetpack4.2)?