realsenseai/librealsense #3166: D435i IMU Scale & Bias Calibration

dorodnic, 2019-01-29:
D435i IMU scale & bias calibration whitepaper is now available:
[RealSense_Depth_D435i_IMU_Calib.pdf](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf)

barnjamin, 2019-01-29:
I've noticed that subsequent runs that the raw norm stays around 9.67 while the corrected is very close to 9.806.  Haven't had time to chase the reason down yet but is this expected?

ev-mp, 2019-01-30:
@barnjamin , the IMU calibration does not modify the raw data generated by D435i.
The IMU readings are adjusted in [librealsense](https://github.com/IntelRealSense/librealsense/blob/development/src/ds5/ds5-options.cpp#L190-L196).

barnjamin, 2019-01-30:
brilliant, this is what i was looking for, thank you @ev-mp

barnjamin, 2019-02-03:
In addition to using the box as a way to keep the camera level, I got my best result using my 3D printer bed after re-leveling it relative to the world.  Turns out my whole house is off level  :laughing:

closed 2019-02-07 by dorodnic