realsenseai/librealsense #3167: single frame can not be aligned

ahuizxc, 2019-01-29:
my librealsense version is 2.17.1 
Here is my code:

rs2::align align_to(RS2_STREAM_COLOR);
auto aligned_depth_frame = depth_frame.apply_filter(align_to);
# depth_frame is rs2::depth_frame from my camera
and i can not get the aligned depth_frame, the aligned_depth_frame is same as depth_frame,
but when i try this code:
rs2::frameset processed = frames.apply_filter(align_to);
auto aligned_depth_frame = processed.get_depth_frame();
and it works well if the frames contain color_frame, so here is my problem:
I think if it is doesn't need to get the color_frame when you want align the depth_frame.

dorodnic, 2019-01-29:
Hi @ahuizxc 
You are right, but how does it affect your application?
What we actually need is the intrinsic / extrinsic data from color frame profile. It is more convenient to have unified API that operates on frames.

vladsterz, 2019-01-30:
Then maybe aligning by yourself would do the job.

dorodnic, 2019-01-30:
The question is why it goes down to 20hz. If its on Intel platform, then our code is not sufficiently optimized and we need to work on that. If you are on a Jetson, we have CUDA optimization that can be enabled. 
On some other low-power platform, alignment itself can be a lot. 
If what slows thing down is the RGB stream, you could configure it to `RS2_FORMAT_YUY2`. This will prevent internal conversion and reduce CPU utilisation.

I was wondering what can you do with just aligned depth, without color?
I mean, most people use `align` to get to an RGBD image and run some algorithms that needs all 4 channels. If you need only depth, why not use the original depth?

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Hi ahuizxc,

Wonder if you have any updates regarding to dorodnic's suggestions and questions.

Thanks!

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi ahuizxc,

Wonder if you still need help on this one?

Thanks!

ahuizxc, 2019-02-13:
> Hi ahuizxc,
> 
> Wonder if you still need help on this one?
> 
> Thanks!

No，I have tried align depth with ssse, but the performance even worse

RealSenseCustomerSupport, 2019-02-13:
----------------------------------------------

Hi ahuizxc,

So you do not need any help on this one?
And by following what what dorodnic suggested, you got worse performance? Would you like to share more details?

Thanks!

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi ahuizxc,

Any updates?

Thanks!

closed 2019-02-25 by dorodnic

ljklxyz, 2019-11-28:
i think the request is valid in my application as well
- i need to calibration extrinsic using rgb stream
- i only need to turn on depth stream for my application to limit bandwidth and decrease latency
- so I need the aligned depth as my camera is calibrated wrt to the color frame

though i entirely understand the rationale of the function design here, and it is possible for me to align them outside librealsense, but still I would prefer librealsense to do so or at least give some warnings when align.proccess(frameset) is called so that we know the depth i get is actually not aligned.