realsenseai/librealsense #3168: Exporting depth frame to grayscale[python wrapper]

abhay44, 2019-01-29:
Hi,
I am trying to export the depth frame to gray scale which looks like this:

[image]

I already checked out this [link](https://github.com/IntelRealSense/librealsense/issues/1831)(I need a Python code since it would be easier to integrate my project). I tried the following, but it's applying **jet colormap**. 

`colorizer = rs.colorizer()
colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())`

How to get the same depth frame as the one in the realsense-viewer? Also, how to filter out the noise to get a smoother image?
Thanks!

lramati, 2019-01-29:
The colorizer's constructor takes an argument for which color scheme to
apply. Unfortunately, I don't recall which value is which.

lramati, 2019-01-29:
The python module supports the built in help() command. Additionally, as
the python API is one-to-one with the C++ API, the C++ examples can easily
be translated to Python

abhay44, 2019-01-30:
@lramati Thank you. I can find the values in colorizer.cpp [here](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/colorizer.cpp#L141)

closed 2019-01-30 by abhay44