realsenseai/librealsense #3170: Unity Multithread mode - ArgumentException: GCHandle value belongs to a different domain

wodndb, 2019-01-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Unity C#                          | 
| Segment			|  others                          | 

### Issue Description

Argument Exception is randomly occurred when streaming is stopped.

Especially, When I use VideoFrame.CopyTo function in Update function, That problem is occurred more frequently.

However, this problem is only occurred when RsDevice is set to multithread mode. When I change multithread mode to unity thread mode, argument exception isn't occurred. I don't know why...

```
ArgumentException: GCHandle value belongs to a different domain
System.Runtime.InteropServices.GCHandle.op_Explicit (System.IntPtr value) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.InteropServices.GCHandle.FromIntPtr (System.IntPtr value) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
Intel.RealSense.CustomProcessingBlock.ProcessingBlockFrameCallback (System.IntPtr f, System.IntPtr u) (at <4fd4dfd1857e4955a500c05c5081d4c2>:0)
Intel.RealSense.FrameSource.FrameReady (System.IntPtr ptr) (at <4fd4dfd1857e4955a500c05c5081d4c2>:0)
Intel.RealSense.FrameSource.FrameReady (Intel.RealSense.Frame f) (at <4fd4dfd1857e4955a500c05c5081d4c2>:0)
RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:103)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:110)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame()
Intel.RealSense.CustomProcessingBlock:ProcessFrame(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:146)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:159)
System.Threading.ThreadHelper:ThreadStart()
```

ogoshen, 2019-02-19:
This should have been addressed, see [this](https://github.com/IntelRealSense/librealsense/blob/8cd5509f1606446ffe065b477694ae9a5ed18e26/wrappers/csharp/Intel.RealSense/Processing/CustomProcessingBlock.cs#L34).
Happens when a managed callback is called from unmanaged code but the handle has been freed.

Not sure how `VideoFrame.CopyTo` is related here, we're already using something similar in `RsStreamTextureRenderer.Update` and `RsPointCloudRenderer.Update` without issues... 

If this is still an issue I'll need to see more code to be of assistance.

wodndb, 2019-02-20:
During 3 weeks I upgraded unity version and changed camera to D415. I try to use multithread mode in unity today, and It works well. Now It's not issue. I appreciate your reply.

wodndb, 2019-02-20:
Same problem is occurred, I'm going to find what is problem in my sourcecode.

wodndb, 2019-02-20:
I want to get raw data of RGB, Depth, and Point Cloud because of image processing in Unity. So I merged '''RsStreamTextureRenderer.Update''' and '''RsPointCloudRenderer.Update'''. Then I copied vertices, RGB data, Depth data.

wodndb, 2019-07-29:
I updated realsense SDK to 2.24.0. The problem is not occurred now.

closed 2019-07-29 by wodndb