realsenseai/librealsense #3171: How to set D435 configuration to get stable depth data.

yeongwoonIm, 2019-01-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | 05.08.15.00  | 
| Operating System & Version |   {Win (8.1)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0 (2018-04-17)                          | 
| Language                            |  C#                     | 
| Segment			|                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

How to set D435 configuration to get stable depth data.

depth data is very unstable to express object.. like a box.
[https://blog.naver.com/PostView.nhn?blogId=ywim0512&Redirect=View&logNo=221454880703&categoryNo=1&isAfterWrite=true&isMrblogPost=false&isHappyBeanLeverage=true&contentLength=4365](url)

would you guys help me..??

MartyG-RealSense, 2019-01-30:
Intel's illustrated guide to camera tuning is an excellent reference for getting better quality results by adjusting camera settings.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf?language=en_US

Could you tell us more please about why you believe that your depth data is unstable.  The depth and IR images look good quality in my opinion. 

For example, are the depth measurement values of the box clearly wrong?

If you are concerned about the multiple dots on the IR image, this is the camera's IR projector projecting a semi-random pattern of dots onto the boxes.  This can help the camera to do analysis of objects with low texture detail such as doors and walls.  

If you do not want the dot pattern on the image, it can be turned off in the RealSense Viewer by de-selecting the 'Emitter Enabled' option.  Left-clicking on the blue box beside this option will turn its color from blue to black.  When it is black, it means that the emitter is turned off.

[image]

closed 2019-02-15 by RealSenseCustomerSupport

yeongwoonIm, 2019-03-14:
Why I thought the Depth data is unsable is the edge of box is moving