realsenseai/librealsense #3173: CV2 Error: Can’t Open File in Function ‘ReadProtoFromTextFile’

ShadowHunter0510, 2019-01-30:
### Issue Description
I have been trying to run the sample code from distance_to_object.ipynb but I have been encountering a problem when initializing the variable net. I’ve been receiving a cv2 error because the program can’t open the two files in the function. I’ve downloaded both files from a GitHub repository and the files are located in the stated locations. How should I fix this issue?
[image]

dorodnic, 2019-01-30:
Maybe mismatching opencv version? We tested it with OpenCV 3 (there is an on-going PR #3091 for OpenCV 4 support, but I haven't tried it yet)
Another reason may be the `~` in file names. Did you try to expand it to something like `/usr/username/home`? 
These are just some initial thoughts, but I hope it can help.

dorodnic, 2019-02-11:
@ShadowHunter0510 - any luck with this?

closed 2019-02-15 by RealSenseCustomerSupport

flavves, 2020-11-02:
> Maybe mismatching opencv version? We tested it with OpenCV 3 (there is an on-going PR #3091 for OpenCV 4 support, but I haven't tried it yet)
> Another reason may be the `~` in file names. Did you try [...]

thats right

eszabo12, 2021-08-11:
I'm still getting this error when the opencv loaded in was explicitly 3.4 as instructed