realsenseai/librealsense #3177: Depth quality tool issue, no point cloud visualization in 3D mode

scipione1, 2019-01-31:
Hi,

I'm trying to display the point cloud of my Intel Realsense D435i with the Depth Quality Tool. However, all I obtain is shown in this figure:

[image]

How do I display the point cloud?

MartyG-RealSense, 2019-01-31:
I spent a hour running tests with the Depth Quality Tool's 3D mode.  I do not usually use the Depth Quality Tool, so this was my first time working in point cloud mode.   Like you, I could not make a point cloud appear no matter what I did, even when the camera was correctly aligned to the wall.

I eventually found that a consistent way to make the point cloud appear was to drop down the Region Of Interest menu and select the '20%' option.  then the point cloud appeared immediately.  I repeatedly closed the tool, re-launched it and repeated the action, and it worked every time for me.

[image]

ev-mp, 2019-01-31:
@scipione1 , the issue you've encountered should be resolved by [PR3130](https://github.com/IntelRealSense/librealsense/pull/3130). It is scheduled for the next increment.
Please check and update.

dorodnic, 2019-02-10:
Now part of [2.18.1](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2181)

closed 2019-02-10 by dorodnic