realsenseai/librealsense #3179: How to access/change frame metadata parameters using python?

amaanda, 2019-01-31:
I have already read the documentation and the examples. The problem I am trying to solve is timestamp synchronization. According to a few issues I found on the repository, the Depth vs Color/IR timestamp desynchronization could be caused by auto-exposure parameters. I also checked the `frames.get_frame_timestamp_domain()` parameter and it returned `system_time`.
So I wanted to manually access and change these parameters, to see if this can solve the problems I'm facing.

Thanks in advance.

amaanda, 2019-01-31:
Both `depth_frame = frames.get_depth_frame()` and `ir_frame = frames.get_infrared_frame()` return `system_time`. So I'm guessing they should be synchronized?

dorodnic, 2019-02-03:
Hi @amaanda 
I'd recommend first trying to address the source of the problem - 
1. [Enable hardware timestamps on Windows](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows) / [Enable hardware timestamp on Linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)
2. You can set Auto-Exposure Priority control to prevent RGB from reducing FPS

As for modifying frame data - this is something we do not enable in the API. Since frame object is shared (you can have many frame objects pointing to same frame data) modifying it in one place could have unexpected side-effects in another part of the program. 

Perhaps you could store the extra information in some external data-structure?

amaanda, 2019-02-04:
I followed the steps provided in the link you sent, and apparently the problem is now solved. 
Thanks!

dorodnic, 2019-02-06:
I think you can't, but you can get the list of enum values using standard python tools, something like:
```py
enum_list = list(map(int, rs.frame_metadata))
```

amaanda, 2019-02-06:
Ah ok. Thanks!

amaanda, 2019-02-06:
I tried it and got the following error: module 'pyrealsense2' has no attribute 'frame_metadata'.

lramati, 2019-02-06:
Try `frame_metadata_value` instead?

RealSenseCustomerSupport, 2019-02-10:
----------------------------------------------

Hi amaanda,

You also can directly query each metadata parameter. For example, getting frame counter with following code snippet:
counter = rs.frame.get_frame_metadata(frame, rs.frame_metadata_value.frame_counter)

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi amaanda,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-02-20 by RealSenseCustomerSupport

amaanda, 2019-02-22:
counter = frames.get_frame_metadata(frames, rs.frame_metadata_value.frame_counter)
TypeError: get_frame_metadata(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.frame, frame_metadata: pyrealsense2.pyrealsense2.frame_metadata_value) -> int

amaanda, 2019-02-22:
I tried using the function in a frameset and also in a depth frame. Both didn't work.

lramati, 2019-02-22:
Remove the frames argument to the function, it shouldn't be there. The call
should be `frames.get_frame_metadata(rs.frame_metadata_value.frame_counter)`
As you can see from the error, the argument of type frame is the `self`
argument, meaning it's the one before the function name, and shouldn't be
inside the parentheses

amaanda, 2019-02-22:
I ran `counter = frames.get_frame_metadata(rs.frame_metadata_value.frame_counter)` and the counter returned 788529152 in a 24 seconds frame. The frame_counter is not returning what it should, I have a 30FPS 24 seconds long .bag, with Infrared 1, Infrared 2 and Depth channel.