realsenseai/librealsense #3181: Efficient points frame computation

dmandrioli, 2019-01-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                          | 
| Language                            |  python                          | 
| Segment			|  Other                          | 

### Issue Description
I would like to create a numpy array mapping each pixel to (x,y,z) world coordinates, like a depth frame with extra (x,y) real world coordinates for each pixel. I did a trivial implementation calling rs2_deproject_pixel_to_point for each pixel but it's slow. Are you aware of any better strategy to do this? Using numpy vectorization or using pointcloud API, I don't know. Python code is welcome.

dmandrioli, 2019-01-31:
Taking inspiration in box_dimensioner_multicam example, this code looks to work:

`   

    def world_points(depth_image, camera_intrinsics):
        [height, width] = depth_image.shape
        nx = np.linspace(0, width - 1, width)
        ny = np.linspace(0, height - 1, height)
        u, v = np.meshgrid(nx, ny)
        x = (u.flatten() - camera_intrinsics.ppx) / camera_intrinsics.fx
        y = (v.flatten() - camera_intrinsics.ppy) / camera_intrinsics.fy
        z = depth_image.flatten() / 1000;
        x = np.multiply(x, z)
        y = np.multiply(y, z)
        return np.reshape(np.dstack(x, y, z), (camera_intrinsics.height, camera_intrinsics.width, 3))

`

closed 2019-01-31 by dmandrioli

dorodnic, 2019-02-03:
Hi @dmandrioli 
Glad you found a solution. An alternative is to use `rs.pointcloud` class. 
It would be interesting to compare how it performs vs the numpy approach.