realsenseai/librealsense #3183: [Android + D435] The connected device doesn't support depth streaming!

oishi89, 2019-01-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |   Android 8.1.0  | 
| Platform                                 | Nexus 6  |
| SDK Version                            |  2.18.0 | 
| Language                            |    C/C++                   | 
| Segment			|  Smartphone                           | 

### Issue Description
I'm trying to run a sample depth app on my android device ./rs-depth and I got an issue like 
"The connected device doesn't support depth streaming!"

dorodnic, 2019-02-03:
Hi @oishi89 
Are you able to run `rs-enumerate-devices`?

oishi89, 2019-02-04:
Hi @dorodnic 
Yes, It's able to.

dorodnic, 2019-02-05:
Hi @oishi89 
This error is reported because the device is connected via USB2 and thus does not support the depth resolution hard-coded in this example. You can modify [rs-depth.c:21-23](https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c#L21-L23) and replace 640x480x30 with any of the Depth combinations reported by `rs-enumerate-devices`

dorodnic, 2019-02-05:
Also, if you perform a [firmware update](https://downloadcenter.intel.com/download/28377/Latest-Firmware-for-Intel-RealSense-D400-Product-Family) you will get more options for USB2. (note that librealsense 2.18.0 is not working well with firmware 5.11.1, but any recent firmware will do)

oishi89, 2019-02-06:
@dorodnic Thanks for your support. It's working now but sometimes it's been disconnected in between. Hopefully, updating firmware might help so what firmware version you recommend?

dorodnic, 2019-02-07:
The link above is to 5.10.13, I think it should be good for you.
Also consider checking out #3083 which offers a new approach to Android development (does not require rooting)

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@oishi89 Any update from your side? Does it get through? Looking forward to your update. Thanks!

oishi89, 2019-02-15:
@RealSenseCustomerSupport I haven't had a chance to work on that yet. I will update you guys as soon as I can. Thanks for your support

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

@oishi89 Any update now? Thanks!

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@oishi89 If no other questions, will close the ticket accordingly.

closed 2019-03-04 by RealSenseCustomerSupport